ros-drivers/ros2_ouster_drivers #34: Update drivers for multiple time sources

SteveMacenski, 2020-04-09:
Port this over: https://github.com/ouster-lidar/ouster_example/pull/97

tpanzarella, 2020-04-10:
Hey @SteveMacenski if this work has not yet started, I'm happy to do it (you can "assign" me) as it is relevant to what I am currently working on. I also have an OS1-16 to test with in my quarantined WFH setup. LMK.

SteveMacenski, 2020-04-10:
That would be great! Thanks.

Are you also generally interested in being a maintainer on this?

You clearly understand what's going on here and a prime user. I'd be more than happy to have you - just add your name to the `package.xml` in the next PR and I'll work on getting you admin.

tpanzarella, 2020-04-10:
Sure. Thanks Steve. Happy to join on as a maintainer/admin.

I'll get a PR together for this issue in the coming days. I'm currently deep on a few other things that I do not want to context switch out of, but this is squarely on my radar (per my earlier comment).

SteveMacenski, 2020-04-10:
Sounds good. This also resolves one of the bullets in #16 to allow for local, ROS, and synced time.

PR #36 (2020-04-13): Support for accessing and mutating the timestamp mode on the LiDAR
The on-LiDAR `timestamp_mode` can now be set in the config file. Additionally,
its state is fedback in the `GetMetadata` service.

I HIL-tested these changes via visual inspection and everything seems to
be working correctly. I see that in #10 there is a plan add some
automated testing. I think this is a *very good* idea. I'll add some comments
to that issue shortly.

Closes: #34
commit bcd032a4 (2020-04-13): Support for accessing and mutating the timestamp mode on the LiDAR (#36)

* Support for accessing and mutating the timestamp mode on the LiDAR

The on-LiDAR `timestamp_mode` can now be set in the config file. Additionally,
its state is fedback in the `GetMetadata` service.

I HIL-tested these changes via visual inspection and everything seems to
be working correctly. I see that in #10 there is a plan add some
automated testing. I think this is a *very good* idea. I'll add some comments
to that issue shortly.

Closes: #34

* Adds TIME_FROM_ROS_RECEPTION as valid timestamp_mode

* When using TIM [... cut]
closed 2020-04-13 by SteveMacenski