ros-drivers/ros2_ouster_drivers #43: Don't use IP addresses in params file

rotu, 2020-05-20:
Right now, the default addresses are hardcoded unicast IPV4 addresses. It would be nice to change this for usability:

1. The lidar advertises itself as `os1-############.local`, where the `#` are the serial number printed on the device. It would be nice to use this hostname instead of the IP address, which may change with a particular hardware setup.

2. The lidar supports more robust destination addresses: multicast with address `255.255.255.255`, `ff02::01` or unicast with `set_udp_dest_auto`.

https://github.com/SteveMacenski/ros2_ouster_drivers/blob/ed188b388e09c9fab54aed21855ba04fcf6c3c08/ros2_ouster/params/os1.yaml#L3-L4

SteveMacenski, 2020-05-20:
On bullet 1, I don't think there's anything stopping you from using that. Have you had a problem with it? If you do, then your problem is with the Ouster SDK because we just feed that string right into it. There's nothing unique about that parameter, its just a string of information, you can stick whatever you want into it. I don't see why it wouldn't work. This is just the default example configuration (which happens to be my functional configuration)

On bullet 2, not sure what's stopping you there

rotu, 2020-05-20:
Nothing is stopping me, except for `set_udp_dest_auto` which would need to be implemented.

I think these are best practices that the example config file should incorporate, since it serves as de facto documentation.

SteveMacenski, 2020-05-20:
Where is `set_udp_dest_auto`? Is that in the latest update to their client?

rotu, 2020-05-21:
> Where is `set_udp_dest_auto`? Is that in the latest update to their client?

`set_udp_dest_auto` is a command introduced in firmware 1.13.0 and documented in the [user guide](https://www.dataspeedinc.com/app/uploads/2019/12/Ouster-OS1-Software-User-Guide.pdf).

> Maybe in the absence of actual documentation, but look at that readme, its _extensive_.

I know! It's AWESOME. But even with that, it's still not obvious [what can go in these items](https://github.com/SteveMacenski/ros2_ouster_drivers/issues/44). Plus I always copy-paste the example config file to get started.

> Also what would you prefer, only have the local address? The same argument can be made the opposite way then. These are identical concepts. the local address maps to an IP.

The local hostname maps to a particular piece of hardware, and it can be determined from a sticker on the unit. The IP address changes whenever the DHCP lease is renewed. Therefore, I prefer the hostname.

Something in the YAML like:
```yaml
# Replace the number with the 12-digit serial number of your OS1 lidar
lidar_ip: os1-990123456789.local
# You could also use its assigned IP address instead, e.g.:
# lidar_ip: 10.5.5.96
```

SteveMacenski, 2020-05-21:
If you wanted to add to the readme documentation that this is an option, I'd merge that, but the default parameter set will remain. I use that as part of the networking examples in the readme about how to setup a quick start connection. If that's the quick start instructions then I want the quickstart parameters to match.

> set_udp_dest_auto is a command introduced in firmware 1.13.0 and documented in the user guide.

PRs are welcome

SteveMacenski, 2020-05-21:
As an aside, there are many situations on a product especially with multiple sensors that you’d rather work with IP addresses. If you have a static set of IPs, you can set all of your front_lidars to an address, back_lidars to an address, .... then as you bootstrap a new robot/car, you’re just installing a sensor from inventory preconfigured at an address and you’re on your way without any unit-specific configurations. Which also makes hot swaps easier.

rotu, 2020-05-21:
> PRs are welcome

I know. I'd love to, but I'm making a conscious effort to try not to hack on things that are off-task.

> As an aside, there are many situations on a product especially with multiple sensors that you’d rather work with IP addresses.

Agreed. That would be the right thing to do if you self-assign the lidar an IP address with `PUT system/network/ipv4/override` or if you create a DHCP rule to assign a fixed IP address based on the MAC address (`bc:0f:a7` = Ouster). But (1) we're using DHCP without such rules (2) I don't think there's a way to set a static address with the TCP interface and it has the downside of making the device unreachable if you screw up.

SteveMacenski, 2020-05-21:
If you're buying a massive order of lidars from a company like ouster, you just tell them what IP to set it up with from the factory. They want to take your money :-) Source: I've done that with other lidar vendors.

tpanzarella, 2020-05-21:
To give the appearance of a static IP address, what we do is hard wire the LiDAR(s) to the computer (i.e., no switches or anything in between the computer and the LiDAR ... this is also for performance). Then for each physical ethernet interface on the computer, we bind a static IP address and make sure that each one is on a separate subnet -- this also has the benefit of making the kernel routing table very easy to read and debug. Then for each subnetwork (i.e., a "dedicated channel" to a LiDAR) we allocate exactly one address in a DHCP pool (obviously, that address is different from the static IP bound to the interface on the host also participating on that subnet). Since the LiDAR is the only device (beside the computer) on the given subnet, it always gets the same address. This insulates us from any device specific configuration and scales nicely as now you map subnetworks to things like "top LiDAR", "rear LiDAR", etc.

rotu, 2020-05-21:
That's a great setup (if you allocate one network per device).

For my setup, I didn't bother with dnsmasq and just plugged the lidar into the nearest ethernet port, which happens to be on a Google WiFi router in my home "office". Using the hostname allows my computer to find that device even if I plug it in elsewhere, and would accomodate additional lidars. It has the downside that, should I need to replace hardware, I would need to change the config.

SteveMacenski, 2020-05-21:
Can we call this ticket closed or rename / focus it on the `set_udp_dest_auto`/

rotu, 2020-05-21:
For using example code, I still think a lidar hostname is more appropriate and user-friendly, even if a field deployment may wish to use IP addresses. Also, I'm open to using a multicast address for the UDP destination, instead of hardcoding.

Submitted a patch because I could no longer resist.

> Can we call this ticket closed or rename / focus it on the `set_udp_dest_auto`?

This ticket should be resolved when those IP addresses are removed or if you decide that one or both IP addresses *do* belong in the config file. How to implement `set_udp_dest_auto` certainly *informs* this ticket (e.g. should that be enabled by setting `computer_ip=""` or by adding a boolean flag?) but I think it's appropriate to call that a separate issue.

SteveMacenski, 2020-05-21:
I also go through very approachable instructions for setting up DHCP on IPs so I think that's also very user friendly. I think this is a case that you prefer to work with the aliases, but I think new users via Ouster's own instructions, my additional thread on them, and 2 production users' experience makes a compelling argument not to change this. I'm making the judgement call not to change this unless Tom disagrees, but it sounds like we're in agreement. 

I added the `set_udp_dest_auto` firmware update feature in this ticket: https://github.com/SteveMacenski/ros2_ouster_drivers/issues/16. But just to set expectations, if this is something you value, I'd recommend implementing it. This is unlikely something that I will do. This will likely only be implemented in my next driver-updates-binge which given the status of my backlog is easily months away.

rotu, 2020-05-22:
> instructions for setting up DHCP on IPs so I think that's also very user friendly.

The whole setup you detail is very fragile, though. You need to change the parameters file if either (1) the DHCP lease expires (e.g. the network is down for an hour) or (2) if the lidar hardware is swapped.

If you're going with IP addresses, ensure they're assigned predictably like @tpanzarella suggests so that both (1) and (2) work seemlessly.

With hostnames, (1) is automatically fixed and you don't need to ensure a dedicated subnet. The downside it that you *can't* fix (2).

SteveMacenski, 2020-05-22:
I understand that you think that the aliases is the right way for your development practices, but I think what Tom and I are trying to say is that we disagree and what you want has drawbacks we don't want to introduce.

Documentation is good, letting people know that they can do this, but these are analog concepts. Saying "hey, you can use the `.local` hostname if you want" is a different thing than "change everything to use it".

rotu, 2020-05-22:
My point is that you already have those drawbacks because of your documented setup process. Changing to hostnames *OR* changing the DHCP instructions would fix this.

> Saying "hey, you can use the .local hostname if you want" is a different thing than "change everything to use it".

Agreed. I think the `.local` hostname is a better place to *start* and a better practice for *development*. When designing the whole system, IP addresses may make more sense.

SteveMacenski, 2020-05-22:
Sure, but this is also the vendor recommendation from their own documentation and in-person demos. Having alignment with the vendor is also valuable.

rotu, 2020-05-22:
I guess? This repository is *the place* to establish best practices for ROS2 and Ouster, where it makes sense to align and where it makes sense to depart from the vendor's more specific recommendations.

I did come across this guide for Ouster and ROS that makes the hostname recommendation but I have no idea where this document came from:

https://levelfivesupplies.com/wp-content/uploads/2019/03/OS-1-User-Guide-Software.pdf

 > NOTE: The sensor hostname does not change when you restart the sensor, whereas IP usually does with a DHCP server. We recommend using hostnames (e.g., os1-991827000891.local) instead of IP addresses for this reason. This functionally achieves the same result as a fixed sensor IP address.

SteveMacenski, 2020-05-22:
I think we can close this then - but follow up on the PR for the specifics in documentation we're already doing over there. Ping if you want it reopened.

closed 2020-05-22 by SteveMacenski

rotu, 2020-05-22:
Okay. Since IP addresses are regarded as a desirable design recommendation, I moved the networking setup to #47

rotu, 2020-05-22:
FYI it's only half-true that IP addresses align better with the vendor recommendations. While the docs always show the lidar sending data by IP address, the TCP connection is always made to the lidar by hostname.