ros-drivers/ros2_ouster_drivers #47: Recommended networking setup does not assign addresses robustly

rotu, 2020-05-22:
The current recommendation on `dnsmasq` setup results in a fragile system since it doesn't result in stable IP addresses.

1. Without hardware changes, if the DHCP lease expires (e.g. the network connection is down for an hour), the configured address is no longer correct.

2. If the hardware is swapped out, the lidar will not be reachable at its configured address without additional action.

Here are some mutually exclusive suggestions for ways to fix this:

* Assign static IP addresses (solves 1, 2 requires additional per-LIDAR setup, Ouster dissuades this)
* Set up dnsmasq so leases are infinite (solves 1, not 2)
* Assign an IP address for the device based on MAC address (solves 1, not 2)
* Constrain the DHCP to assign addresses in a predictable way, emulating static assignment. [A description of one such implementation that also requires restricting the physical network](https://github.com/SteveMacenski/ros2_ouster_drivers/issues/43#issuecomment-632017226). It may be possible to lift some of these restrictions by making dnsmasq use, e.g. a certain pool for certain vendor types. (solves 1 and 2)

SteveMacenski, 2020-05-22:
@tpanzarella is this worth doing? The networking instructions are just from Ouster themselves with more explanation and color. I'm really not unhappy with the state of things right now, obviously anyone using this sensor for their serious networks is going to have preferences for how its setup (static, dynamic, .local, etc) but I don't think that means to change what the vendor's quickstart suggestions layout. 

I'd be OK also adding a subsection explaining out to setup the static IPs.

tpanzarella, 2020-05-22:
> I'd be OK also adding a subsection explaining out to setup the static IPs.

I agree with this approach. @rotu brings up a point that he thinks will better the user experience of the system. He points are valid. However, I agree with you Steve that this is not really a technical problem that requires changing code or config files. I think a balance can be found in updates to the documentation.

closed 2020-06-15 by SteveMacenski

RFRIEDM-Trimble, 2021-06-02:
Has anyone considered hostname resolution, IPv6 link local addressing, and finally `set_udp_dest_auto`. If that all worked, it no longer requires configuring static IP's on the sensors and mucking with network settings on the host. IPv6 link local addresses come up automatically by default on both the sensor and a linux host.

PR #54 (2020-06-15): adding .local options and broadcast
To supersede #45 and close #47