ros-drivers/ros2_ouster_drivers #60: /scan topic always has 0 values for range and intensity

Mohit-Ak, 2020-10-23:
Issue: /scan(Laser Scan) topic always has zero values in range and intensity.
OS: Ubuntu 20.04
ROS2: Foxy
Sensor: OS-32
Steps to reproduce:

1. Clone the repository inside src of a ROS2 workspace
```git clone -b foxy-devel https://github.com/ros-drivers/ros2_ouster_drivers.git```
2. Follow the steps mentioned in the readme file to set up the [sensor](https://github.com/ros-drivers/ros2_ouster_drivers)
3. The Mode that was tried where 1024x10 or 512x10. These were set in the ```os1.yaml``` file.
4. Ensure that SCAN is present in os1_proc_mask ```os1_proc_mask: IMG|PCL|IMU|SCAN``` 
5. Run ```ros2 launch ros2_ouster os1_launch.py```
6. Open RVIZ and add set the frame to ```laser_sensor_frame```
7. Add Pointcloud and Laser scan topic to the display panel.
See PointCloud in ```/points``` topic.
[image]

See No Laserscan in ```/scan``` topic.
[image]

Sample ROS topic echo of ```/scan```
```
header:
  stamp:
    sec: 1812
    nanosec: 946384100
  frame_id: laser_data_frame
angle_min: -3.1415927410125732
angle_max: 3.1415927410125732
angle_increment: 0.006135923322290182
time_increment: 9.765625145519152e-05
scan_time: 0.10000000149011612
range_min: 0.10000000149011612
range_max: 120.0
ranges:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
<More zeros>
intensities:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
<More zeros>
```

bwr-eng, 2020-10-25:
I have the same problem. I can't view also the point cloud on Rviz but with `ros2 topic echo` I can see it.

SteveMacenski, 2020-10-26:
@bwr-eng clearly your issue is QoS if you can't even get any data out of it, please refrain from derailing over tickets.

@Mohit-Ak I am unable to reproduce.

Mohit-Ak, 2020-10-27:
Do you need anything else from my side to reproduce?

Mohit-Ak, 2020-10-27:
> @bwr-eng clearly your issue is QoS if you can't even get any data out of it, please refrain from derailing over tickets.
> 
> @Mohit-Ak I am unable to reproduce.

Do you need anything else from my side to reproduce?

SteveMacenski, 2020-10-27:
Well at the moment there's no action to be taken since we can't reproduce.

Mohit-Ak, 2021-08-04:
The issue got resolved in the new 2.x firmware + ROS driver update.

closed 2021-08-04 by Mohit-Ak