ros-drivers/ros2_ouster_drivers #61: Rviz2 show 0 messages of topic /points

bwr-eng, 2020-10-26:
Hi,

I can see the data on `ros2 topic echo /points` but in `Rviz2` I can't see it. Rviz notified that `0 messages arrived`.
I tried to reinstall `rviz2` but hadn't solved. 
If I opened it with `rviz` of ROS1 (by the bridge pkg) I can see the point clouds.
Do you know what is the reason for that problem?

Aviad

tpanzarella, 2020-10-26:
You may want to check the Quality of Service drop down for the topic on the rviz side. `/points` is published with the [sensor data QoS](https://github.com/ros2/rmw/blob/foxy/rmw/include/rmw/qos_profiles.h#L25) so make sure the rviz side is [compatible](https://index.ros.org/doc/ros2/Concepts/About-Quality-of-Service-Settings/#qos-compatibilities).

closed 2020-12-03 by SteveMacenski

RFRIEDM-Trimble, 2021-06-03:
For anyone curious, here's my settings - System Default QOS. You also need the frames to match. 
[image]