ros-drivers/ros2_ouster_drivers #64: Request to give Main ROS2 Master branch support.

Mohit-Ak, 2020-12-16:
Currently, I see that the last code commit for the driver is a few months old and I believe it is not much of an effort to maintain a ROS2 Master branch that can compile with the latest ROS2. I would have helped but I am not very good at writing C++ drivers for ROS2.

OS: Ubuntu 20.04
ROS2: Master Branch
Sensor: OS-32

**Steps to reproduce:**
- Build ROS2 from source selecting the Master branch.
[https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/
)
- ```git clone https://github.com/ros-drivers/ros2_ouster_drivers.git```
- Ouster_msgs compiles successfully ```colcon build --symlink-install --packages-select ouster_msgs```
-  ros2_ouster throws errors ```colcon build --symlink-install --packages-select ros2_ouster```
```
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:61:5: error: ‘value_to_yaml’ was not declared in this scope
   61 |     value_to_yaml(msg.angle_increment, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:71:5: error: ‘value_to_yaml’ was not declared in this scope
   71 |     value_to_yaml(msg.time_increment, out);
      |     ^~~~~~~~~~~~~
In file included from /home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/laser_scan.hpp:9,
                 from /home/bautomation/ba_pontoon_ros2/src/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:31,
                 from /home/bautomation/ba_pontoon_ros2/src/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/ouster_driver.hpp:21,
                 from /home/bautomation/ba_pontoon_ros2/src/ros2_ouster_drivers/ros2_ouster/src/ouster_driver.cpp:21:
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const LaserScan&, std::ostream&, size_t)’:
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:41:5: error: ‘value_to_yaml’ was not declared in this scope
   41 |     value_to_yaml(msg.angle_min, out);
      |     ^~~~~~~~~~~~~
In file included from /home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform__traits.hpp:17,
                 from /home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform_stamped__traits.hpp:19,
                 from /home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/transform_stamped.hpp:9,
                 from /home/bautomation/ba_pontoon_ros2/src/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:29,
                 from /home/bautomation/ba_pontoon_ros2/src/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/ouster_driver.hpp:21,
                 from /home/bautomation/ba_pontoon_ros2/src/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/driver_types.hpp:17,
                 from /home/bautomation/ba_pontoon_ros2/src/ros2_ouster_drivers/ros2_ouster/src/driver_types.cpp:18:
/home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const Vector3&, std::ostream&, size_t)’:
/home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:28:5: error: ‘value_to_yaml’ was not declared in this scope
   28 |     value_to_yaml(msg.x, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:81:5: error: ‘value_to_yaml’ was not declared in this scope
   81 |     value_to_yaml(msg.scan_time, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:51:5: error: ‘value_to_yaml’ was not declared in this scope
   51 |     value_to_yaml(msg.angle_max, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:38:5: error: ‘value_to_yaml’ was not declared in this scope
   38 |     value_to_yaml(msg.y, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:91:5: error: ‘value_to_yaml’ was not declared in this scope
   91 |     value_to_yaml(msg.range_min, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:61:5: error: ‘value_to_yaml’ was not declared in this scope
   61 |     value_to_yaml(msg.angle_increment, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:101:5: error: ‘value_to_yaml’ was not declared in this scope
  101 |     value_to_yaml(msg.range_max, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:48:5: error: ‘value_to_yaml’ was not declared in this scope
   48 |     value_to_yaml(msg.z, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:71:5: error: ‘value_to_yaml’ was not declared in this scope
   71 |     value_to_yaml(msg.time_increment, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ros_ws_foxy/ros2_ws/install/sensor_msgs/include/sensor_msgs/msg/detail/laser_scan__traits.hpp:119:9: error: ‘value_to_yaml’ was not declared in this scope
  119 |         value_to_yaml(item, out);
      |         ^~~~~~~~~~~~~
In file included from /home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform__traits.hpp:19,
                 from /home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform_stamped__traits.hpp:19,
                 from /home/bautomation/ros_ws_foxy/ros2_ws/install/geometry_msgs/include/geometry_msgs/msg/transform_stamped.hpp:9,
                 from /home/bautomation/ba_pontoon_ros2/src/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:29,
[... 265 lines cut ...]
/home/bautomation/ba_pontoon_ros2/install/ouster_msgs/include/ouster_msgs/msg/detail/metadata__traits.hpp:106:9: error: ‘value_to_yaml’ was not declared in this scope
  106 |         value_to_yaml(item, out);
      |         ^~~~~~~~~~~~~
/home/bautomation/ba_pontoon_ros2/install/ouster_msgs/include/ouster_msgs/msg/detail/metadata__traits.hpp:126:9: error: ‘value_to_yaml’ was not declared in this scope
  126 |         value_to_yaml(item, out);
      |         ^~~~~~~~~~~~~
/home/bautomation/ba_pontoon_ros2/install/ouster_msgs/include/ouster_msgs/msg/detail/metadata__traits.hpp:138:5: error: ‘value_to_yaml’ was not declared in this scope
  138 |     value_to_yaml(msg.serial_no, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ba_pontoon_ros2/install/ouster_msgs/include/ouster_msgs/msg/detail/metadata__traits.hpp:148:5: error: ‘value_to_yaml’ was not declared in this scope
  148 |     value_to_yaml(msg.firmware_rev, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ba_pontoon_ros2/install/ouster_msgs/include/ouster_msgs/msg/detail/metadata__traits.hpp:158:5: error: ‘value_to_yaml’ was not declared in this scope
  158 |     value_to_yaml(msg.imu_port, out);
      |     ^~~~~~~~~~~~~
/home/bautomation/ba_pontoon_ros2/install/ouster_msgs/include/ouster_msgs/msg/detail/metadata__traits.hpp:168:5: error: ‘value_to_yaml’ was not declared in this scope
  168 |     value_to_yaml(msg.lidar_port, out);
      |     ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/ouster_driver_core.dir/build.make:102: CMakeFiles/ouster_driver_core.dir/src/OS1/OS1_sensor.cpp.o] Error 1
make[2]: *** [CMakeFiles/ouster_driver_core.dir/build.make:63: CMakeFiles/ouster_driver_core.dir/src/driver_types.cpp.o] Error 1
make[2]: *** [CMakeFiles/ouster_driver_core.dir/build.make:76: CMakeFiles/ouster_driver_core.dir/src/ouster_driver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:134: CMakeFiles/ouster_driver_core.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```

SteveMacenski, 2020-12-16:
First off, I don't appreciate the insinuations. Not only are they not productive, but they also don't really make me particularly inclined to help fix whatever issues your facing. Especially when those issues aren't caused by this project, as the errors you posted clearly indicates (unless you believe that we're messing around with internal traits of core interfaces for fun).

There hasn't been a commit recently because its been working stably for several months and beyond a few tickets for feature additions, there's not really anything more that's needed for the majority of users as far as ROS2 / master build tasks go.

If you aren't familiar with C++ and unwilling to learn, that's not really our problem.  

---

Now, moving on. Your issue has nothing to do with this driver if you look at the errors. You're having issues with the compiled messages which has nothing to do with this repository, you'd have the same issues elsewhere I bet trying to compile other executibles using those messages. Without knowing alot more about your development environment and source path, there's nothing I can do. 

I just compiled ros2 ouster + messages without any build errors (or warnings, for that matter) against the latest ROS2 master branches. 

Closing ticket, there's no action for maintainers.

closed 2020-12-16 by SteveMacenski

SteveMacenski, 2020-12-16:
If I were to wildly guess, I'd guess that your issue is that you sourced your `/opt/ros/foxy/install/setup.bash` file in one of your workspaces (ros2 master, ouster workspace, etc) in the path which is messing up the paths for interface definitions and that this issue is trivially solved by carefully tracking your source paths.

But please don't assume that you're entitled to personal support when we offer freely available software for everyone to use. Use your best judgement to try to fix the issue and don't assume that every issue you face is our fault.

Mohit-Ak, 2020-12-23:
@SteveMacenski I would like to sincerely apologize if the tone of the ticket was offensive. I honestly did not mean it, as you can see from the title of the ticket itself is "**Request** to give ROS2 Master branch support". Also, in the first sentence, I meant that the code used to compile initially with ROS2 Master but as months went by, felt like the repo got outdated because of the changes to ROS2 base packages. Also, I never demanded free personal support as I did not even know if this is a personal workspace issue or a more generic one. I am just representing a company that has considerably invested in Ouster and this ticket was a consolidation of developer complaints. I believe the company will be more than willing to pay for tech support to Ouster if that is an option.

I just thought, how most other ROS2 packages have a "master" branch that is different from "foxy-devel", should Ouster drivers also follow that approach?

I tried reinstalling ROS2 in a clean slate Ubuntu 20.04 using the script present in the Navigation2 repository but ended up with the same error. All other packages compile fine except for the ```ros2_ouster```. Is there a docker environment you would recommend where I can try the same?

Apologies again @SteveMacenski. I wish I could take back anything hurtful from the above ticket and on a personal scale, I have been a big fan of your work and have been following you since ROSCon 2019.