ros-drivers/ros2_ouster_drivers #65: Unable to connect to the lidar - Address already in use

MaxandreOgeret, 2021-01-06:
Hello, 

I use ros dashing. and my Lidar has the firmware 2.0 and a static ip.
I have configured the yaml file correctly : 

```yaml
ouster_driver:
  ros__parameters:
    lidar_ip: 192.168.3.200
    computer_ip: 192.168.3.100
    lidar_mode: "512x10"
    imu_port: 7503
    lidar_port: 7502
    sensor_frame: laser_sensor_frame
    laser_frame: laser_data_frame
    imu_frame: imu_data_frame
```

But when I try to run the launch file I get this error : 
```
[INFO] [launch]: All log files can be found below /home/student/.ros/log/2021-01-06-15-25-28-707071-HPZ1-14468
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ouster_driver-1]: process started with pid [14478]
[ouster_driver-1] [INFO] [ouster_driver]: Configuring Ouster driver node.
[ouster_driver-1] [INFO] [ouster_driver]: Connecting to sensor at 192.168.3.200.
[ouster_driver-1] [INFO] [ouster_driver]: Broadcasting data from sensor to 192.168.3.100.
[ouster_driver-1] udp bind(): Address already in use
[ouster_driver-1] udp bind(): Address already in use
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_CONFIGURE' for LifecycleNode '/ouster_driver'
[ouster_driver-1] [ERROR] []: Caught exception in callback for transition 10
[ouster_driver-1] [ERROR] []: Original error: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[ouster_driver-1] [WARN] []: Error occurred while doing error handling.
[ouster_driver-1] [ERROR] [ouster_driver]: Handing error in Ouster driver node.
```

Any idae what could be the problem ? It works perfectly with ROS1.

Thanks !

MaxandreOgeret, 2021-01-06:
Error was on me.

closed 2021-01-06 by MaxandreOgeret

Omrisr25, 2021-11-22:
Hey, I have the same problem.. can you tell me what was the error?

MaxandreOgeret, 2021-11-22:
If I remember well, the port was already in use.