ros-drivers/ros2_ouster_drivers #66: Unable to connect

MaxandreOgeret, 2021-01-06:
Hello, 

I use ros dashing. and my Lidar has the firmware 2.0 and a static ip.
I have configured the yaml file correctly : 

```yaml
ouster_driver:
  ros__parameters:
    lidar_ip: 192.168.3.200
    computer_ip: 192.168.3.100
    lidar_mode: "512x10"
    imu_port: 7503
    lidar_port: 7502
    sensor_frame: laser_sensor_frame
    laser_frame: laser_data_frame
    imu_frame: imu_data_frame
```

But when I try to run the launch file I get this error : 
```
[INFO] [launch]: All log files can be found below /home/student/.ros/log/2021-01-06-15-31-51-551814-HPZ1-16973
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ouster_driver-1]: process started with pid [16983]
[ouster_driver-1] [ERROR] []: Caught exception in callback for transition 10
[ouster_driver-1] [ERROR] []: Original error: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[ouster_driver-1] [WARN] []: Error occurred while doing error handling.
[ouster_driver-1] [ERROR] [ouster_driver]: Handing error in Ouster driver node.
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_CONFIGURE' for LifecycleNode '/ouster_driver'
```

Any idea what could be the problem ? It works perfectly with ROS1.

Thanks !

SteveMacenski, 2021-01-06:
Hi,

If you look at this ticket https://github.com/ros-drivers/ros2_ouster_drivers/issues/42 and this comment https://github.com/ros-drivers/ros2_ouster_drivers/issues/63#issuecomment-742027046 you can see that we do not yet support v2 firmware. I'm open to pull requests to enable it but I have not gotten the cycles to do this yet after the holidays. I hope to have time to sit down and do this in the next 2 weeks but I'm also waiting for some resolution on topics discussed with Ouster in this thread https://github.com/ouster-lidar/ouster_example/issues/164

MaxandreOgeret, 2021-01-06:
Hey thanks for your answer.
I am willing to spend some time and make PRs for firmware 2.0 compatibility.

I have an OS-1 128 beams, and I see that the issue on ouster repo is about support of 128 beams sensors. Does it change anything for me ?

I will take the time to go through the code soon.

SteveMacenski, 2021-01-06:
Nope, in fact you require v2.0 to even use a 128 beam, so its totally on topic

MaxandreOgeret, 2021-01-10:
I'm starting to work on it. But I noticed that the dashing version is not up to date and my only interest is for dashing at the moment. What do you advise me to do ?
Why does the dashing version doesn't have the same functionalities ? Is it because something is missing in Dashing ?  Thanks.

SteveMacenski, 2021-01-11:
Because we cannot backport ABI/API breaking changes to already released distributions. If you want all the newest features, you have to use the newer LTS distributions. We can certainly work on getting dashing working on V2 though.

MaxandreOgeret, 2021-01-17:
Hey @SteveMacenski, I created the PR https://github.com/ros-drivers/ros2_ouster_drivers/pull/67.
It's not functionnal yet because i'm stuck with the processors but I'm sure most of the job is done. Could you give me a hand in this ?

Thanks !

MaxandreOgeret, 2021-01-22:
@SteveMacenski No answer, so I guess it's not worth spending more time on this ?

SteveMacenski, 2021-01-22:
It's a very large PR that I haven't had time to review yet. There's enough changes it would take me a couple of hours to really dissect it all (to find some subtle issue why its connecting but not streaming) - typically we like smaller PRs and there's some changes in structs that I didn't expect to be made that I have to track down as well.

MaxandreOgeret, 2021-01-22:
Thanks for your answer. Is it worth it for me to spend more time on it ?

SteveMacenski, 2021-01-22:
I'd continue to try to get it to work since that's something you need, but I wouldn't do anything just for the sake of the PR until I have a chance to look over it.

MaxandreOgeret, 2021-01-22:
We will survive using the ROS1 bridge. Tag me once you had a look and I'll continue with pleasure.

SteveMacenski, 2021-01-28:
@MaxandreOgeret I took a look - I would recommend sanity checking that lidar / IMU packet size being set correctly. If that isn't right, it would result in the behavior you've described where there's no data coming out because the packets are messed up.

Can you tell me where you tracked the data down to? Where was it that things stopped working?

closed 2021-05-04 by SteveMacenski

mnissov, 2021-10-19:
Has this fix been shipped for the `foxy-devel` branch? Experiencing identical problems with an OS0-128

mnissov, 2021-10-19:
Yeah I think you're right, do you recall any of the code being particular to ros2, I assume this means the rolling distro, wrt to foxy?

PR #67 (2021-05-04): Firmware 2.0 compatibility
**Big thanks to Ouster for lending me a sensor to continue this development !**

fixes #66 

Brought client directory from https://github.com/ouster-lidar/ouster_example
This PR still needs a lot of cleaning.

**Include**
 - ros2_ouster
   -  client
     - client.h
     - types.h
     - version.h
     - impl
       - netcompat.h
       - parsing.h

**src**
 - client
   - client.cpp
   - netcompat.cpp
   - types.cpp

I followed the idea in this comment : https://github.com/ros-drivers/ros2_ouster_drivers/issues/63#issuecomment-742027046. But a lot of changes had to be [... cut]