ros-drivers/ros2_ouster_drivers #70: colcon build fails on Foxy

lukeboi, 2021-05-27:
Hi all, I've been working with this library on ROS2 Dashing/Ubuntu 18 and today updated to Ubuntu 20 with a fresh install of ROS2 Foxy. I cloned this repo and have been unable to get colcon build to succeed. [Here](https://gist.github.com/lukeboi/60d30693ee88022e28f1ebc7ec316931) are the steps I've been taking to reproduce the build failure. Please let me know if there's anything else I can provide or do to get this issue fixed.

Cheers,
Luke

SteveMacenski, 2021-05-27:
You need to use the foxy branch, not ros2

lukeboi, 2021-05-27:
Thanks for the speedy reply! I've tried using the contents of both the foxy-devel branch and ros2 branch, I get the same error on both.

SteveMacenski, 2021-05-27:
You probably have something cached wrong somewhere, I just verified (output of history in my terminal):

```
 2004  mkdir -p hi/src
 2005  cd hi/src/
 2006  git clone git@github.com:ros-drivers/ros2_ouster_drivers.git -b foxy-devel
 2007  cd ../
 2008  source /opt/ros/foxy/setup.bash 
 2009  colcon build 
```

builds fine

lukeboi, 2021-05-28:
This was indeed a cache-related issue. After recompiling and reinstalling pcl the packages build fine. Thanks!

closed 2021-05-28 by lukeboi