ros-drivers/ros2_ouster_drivers #81: When using Cyclone DDS topics aren’t being subscribed

jediofgever, 2021-08-15:
Hi , 

I recently attached an Ouster lidar to a robot running on an Intel NUC. 
With the default rmw of ros foxy, I am able to capture the cloud and view it in RVIZ. I want however process the cloud on another machine, the machine and intel NUC are on same network, all topics and nodes are discoverable mutually between these two. 

When view the cloud on rviz at remote machine, the cloud reaches partially. I thought that to change deadlifts DDS to cyclone time see if it can handle this problem. However when rmw is set to Cyclone DDS, none of the topics published by the driver are subscribe-able, I checked all Qos compatibilities, ros2 topic echo just hangs without saying much. 

Any recommendations ? 
Has someone run this driver in a similar setup or non-default DDS ?

SteveMacenski, 2021-09-16:
Is this still relevant?

jediofgever, 2021-09-22:
It was, could not make the driver work when cyclone DDs was selected as default rmw. Maybe it’s my own configuration settings but not sure, so that’s why wanted to see whether anyone tested the driver with cyclone DDs