ros-drivers/ros2_ouster_drivers #83: Failed to create connection to lidar.

bariskplgz, 2021-09-22:
Hi,

I get such an error when I want to run it after completing the installation steps.

[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ouster_driver-1]: process started with pid [129885]
[ouster_driver-1] [INFO] [1632326990.557586247] [ouster_driver]: Configuring Ouster driver node.
[ouster_driver-1] [INFO] [1632326990.557844659] [ouster_driver]: Connecting to sensor at 192.168.1.8.
[ouster_driver-1] [INFO] [1632326990.557915325] [ouster_driver]: Sending data from sensor to 192.168.1.102.
[ouster_driver-1] [FATAL] [1632326991.581233097] [ouster_driver]: Exception thrown: (Failed to create connection to lidar.)

SteveMacenski, 2021-09-25:
Well, is there a lidar at that IP address?

bariskplgz, 2021-09-28:
Yes ip address has a lidar.

DanielDowling1, 2021-11-03:
Having same issue. Can't figure out what I'm doing wrong.
[image]

DanielDowling1, 2021-11-03:
Getting these packets via Wireshark.
[image]

DanielDowling1, 2021-11-03:
Never mind. I seem to have fixed it by connecting it to the LAN instead of directly to my machine and updating the firmware.

bariskplgz, 2021-11-05:
Which firmware version did you upgrade to? My firmware version is v2.1.1

DanielDowling1, 2021-11-08:
I've upgraded to firmware version 2.1.3.

closed 2022-04-27 by SteveMacenski