ros-drivers/ros2_ouster_drivers #84: Frame transformations different from Ouster Software Manual

Gaurav-Kapoor-InMach, 2021-09-29:
Hello
I just checked that the frame transformations on the tf_static topic are different from the transformations given in the Ouster Software Manual. This is the output if the /tf_static topic which is published just once when we start this driver - 

transforms:
- header:
    stamp:
      sec: 1632922020
      nanosec: 663394972
    frame_id: laser_sensor_frame
  child_frame_id: imu_data_frame
  transform:
    translation:
      x: 0.0
      y: 0.0
      z: 0.0
    rotation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
- header:
    stamp:
      sec: 1632922020
      nanosec: 663400150
    frame_id: laser_sensor_frame
  child_frame_id: laser_data_frame
  transform:
    translation:
      x: 0.0
      y: 0.0
      z: 0.0
    rotation:
      x: 0.0
      y: 0.0
      z: 1.0
      w: 0.0
---

When I visualized the data in RViz I saw that the laser_sensor frame is rotated by 180 degrees from the RViz axes. The laser_data_frame in RViz with respect to the real Ouster 0S1-128 Sensor was like this - 
[image]
Which is completely different from the frame transformations given in the manual. So I want to know the correct transformations of the laser_data_frame and imu_data_frame with respect to the laser_sensor_frame and also the correct position and orientation of the laser_sensor_frame with respect to the real Ouster Sensor. 

Thanks.

Best regards
Gaurav

mnissov, 2021-10-20:
I think the rotation part is actually correct, in the [software manual](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2.1.x.pdf) they define three frames: sensor, lidar, and IMU. I imagine these correspond to the `laser_sensor_frame`, `laser_data_frame`, and `imu_data_frame` respectively.

That being said, there is an error in the translation portion of the transformations as the sensors each carry a certain vertial offset between frames. This is illustrated here
[image]
[image]

Aposhian, 2023-05-08:
For reference, it looks like there is a bug in this driver where the driver stamps pointclouds with the `lidar_frame`, but it actually applies the transformation to put them in the sensor frame. See resolution at: https://github.com/ouster-lidar/ouster-ros/issues/33

ga58lar, 2024-04-03:
I just wanted to open an issue with the same error. Is there a reason this is still open? 
@goksanisil23 thanks a lot for the fix!