ros-drivers/ros2_ouster_drivers #89: Pointcloud2 Message frequency drops and data is lost

DavidSku, 2021-12-15:
Hello, I use the driver under ROS2 Foxy for the Ouster OS0-64 sensor. Basically the driver works. Sometimes the frequency with which the Pointcloud2 message is sent is reduced. Nominal this should be 10 Hz in my configuration. If I monitor the frequency with "ros2 topic hz" it sometimes drops to 7-8 Hz. Apparently some messages are lost. For Quality of Service (QoS) I have set "Sensor_QoS". It also looks like the frequency gets lower the more subscribers are active on the Pointcloud2 message.

Do you have any idea what the problem could be?

bertaveira, 2022-01-19:
Same thing happens to me with a OS1-64 with firmware v2.1.3 on foxy

nrabany, 2022-07-01:
Hello, you can increase the socket buffers size to solve this prolem. It is explained here:
[(https://fast-dds.docs.eprosima.com/en/latest/fastdds/use_cases/large_data/large_data.html)]

SteveMacenski, 2022-07-08:
Correct, this is not a driver issue, this is a bandwidth on DDS issue

closed 2022-07-08 by SteveMacenski