ros-drivers/ros2_ouster_drivers #90: Eigen error when starting

bertaveira, 2022-01-19:
From what I have been reading my sensor should not work with the foxy-devel branch but it does although it flickers a bit on the left side of the sensor. 

To check if this was fixed in the newer versions I tried to run the galactic version and the version in the ros2 branch (on a machine with galactic installed) but both gave me the same error:

```
[INFO] [launch]: All log files can be found below /home/bertaveira/.ros/log/2022-01-19-21-57-23-674269-ubuntu-7377
[INFO] [launch]: Default logging verbosity is set to INFO
1642625843.766833 [0]       ros2: using network interface enp0s5 (udp/192.168.0.110) selected arbitrarily from: enp0s5, docker0, enxa0cec8ea264f
[INFO] [ouster_driver-1]: process started with pid [7388]
[ouster_driver-1] 1642625843.840869 [0] ouster_dri: using network interface enp0s5 (udp/192.168.0.110) selected arbitrarily from: enp0s5, docker0, enxa0cec8ea264f
[ouster_driver-1] [INFO] [1642625844.036431420] [ouster_driver]: Configuring Ouster driver node.
[ouster_driver-1] [INFO] [1642625844.037142796] [ouster_driver]: This driver is compatible with sensors running fw 2.x.
[ouster_driver-1] [INFO] [1642625844.037468255] [ouster_driver]: Connecting to sensor at 10.5.5.66.
[ouster_driver-1] [INFO] [1642625844.037495880] [ouster_driver]: Sending data from sensor to 10.5.5.1.
[INFO] [launch.user]: [LifecycleLaunch] Ouster driver node is activating.
[ouster_driver-1] [INFO] [1642625861.387698393] [ouster_driver]: Activating Ouster driver node.
[ouster_driver-1] ouster_driver: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:180: Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType Eigen::DenseCoeffsBase<Derived, 0>::operator()(Eigen::Index) const [with Derived = Eigen::Map<const Eigen::Array<unsigned int, -1, -1, 1, -1, -1> >; Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType = unsigned int; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.
[ERROR] [ouster_driver-1]: process has died [pid 7388, exit code -6, cmd '/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/ros2_ouster/ouster_driver --ros-args -r __node:=ouster_driver -r __ns:=/ --params-file /media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/share/ros2_ouster/params/sensor.yaml'].
[INFO] [launch.user]: [LifecycleLaunch] Ouster driver node is exiting.
```
Sensor: OS1-64
Firmware: v2.1.3
OS: Ubuntu 20.04
ROS: foxy and Galactic

SteveMacenski, 2022-01-19:
Do you have the proper version of eigen installed based on REP 2000?

bertaveira, 2022-01-19:
I don't think so. Even running it on a brand new docker container (ros:galactic) and installing all dependencies gives the same error. Tried both on docker and off, both in arm and amd64 and I always get the same error which is weird

bertaveira, 2022-01-20:
I tried in both architechtures in two different PCs and both in docker and outside. 
So trying on a docker container I did it like this:
```
docker run -it --net=host ros:galactic
```
Then git clone this repo and checkout to the branch `galactic`. After which I have installed the dependencies:
```
apt update && apt install libpcl-dev ros-galactic-perception-pcl
```
Then I changed the IP of the `sensor.yml` file to my lidar IP and build it:
```
colcon build
source install/setup.bash
ros2 launch ros2_ouster os1_launch.py
```

I get the error described in this issue ticket. I get this error despite how I try to do it or whichever branch I try it except for the `foxy-devel` branch which for some reason works but with some blackouts and flickering.

bertaveira, 2022-01-20:
Ran it in debug mode and here is the output from beginning to the crash:
```
ros2 launch ros2_ouster os1_launch.py 
[INFO] [launch]: All log files can be found below /home/bertaveira/.ros/log/2022-01-20-20-08-53-658430-ubuntu-221276
[INFO] [launch]: Default logging verbosity is set to INFO
1642705733.776111 [0]       ros2: using network interface enp0s5 (udp/192.168.0.110) selected arbitrarily from: enp0s5, docker0, enxa0cec8ea264f
[INFO] [ouster_driver-1]: process started with pid [221287]
[ouster_driver-1] [DEBUG] [1642705733.886581062] [rclcpp]: signal handler installed
[ouster_driver-1] [DEBUG] [1642705733.886609520] [rclcpp]: deferred_signal_handler(): waiting for SIGINT or uninstall
[ouster_driver-1] [DEBUG] [1642705733.888535313] [rcl]: Initializing node 'OusterDriver' in namespace ''
[ouster_driver-1] [DEBUG] [1642705733.888553521] [rcl]: Using domain ID of '0'
[ouster_driver-1] 1642705733.888935 [0] ouster_dri: using network interface enp0s5 (udp/192.168.0.110) selected arbitrarily from: enp0s5, docker0, enxa0cec8ea264f
[ouster_driver-1] [DEBUG] [1642705733.891150355] [rcl]: Initializing publisher for topic name '/rosout'
[ouster_driver-1] [DEBUG] [1642705733.891198480] [rcl]: Expanded topic name '/rosout'
[ouster_driver-1] [DEBUG] [1642705733.892353939] [rcl]: Publisher initialized
[ouster_driver-1] [DEBUG] [1642705733.892367772] [rcl]: Node initialized
[ouster_driver-1] [DEBUG] [1642705733.894800940] [rcl]: Initializing service for service name 'ouster_driver/get_parameters'
[ouster_driver-1] [DEBUG] [1642705733.894815815] [rcl]: Expanded service name '/ouster_driver/get_parameters'
[ouster_driver-1] [DEBUG] [1642705733.895987940] [rmw_cyclonedds_cpp]: ************ Service Details *********
[ouster_driver-1] [DEBUG] [1642705733.896009940] [rmw_cyclonedds_cpp]: Sub Topic rq/ouster_driver/get_parametersRequest
[ouster_driver-1] [DEBUG] [1642705733.896012398] [rmw_cyclonedds_cpp]: Pub Topic rr/ouster_driver/get_parametersReply
[ouster_driver-1] [DEBUG] [1642705733.896038773] [rmw_cyclonedds_cpp]: ***********
[ouster_driver-1] [DEBUG] [1642705733.896145732] [rcl]: Service initialized
[ouster_driver-1] [DEBUG] [1642705733.896179857] [rcl]: Initializing service for service name 'ouster_driver/get_parameter_types'
[ouster_driver-1] [DEBUG] [1642705733.896187065] [rcl]: Expanded service name '/ouster_driver/get_parameter_types'
[ouster_driver-1] [DEBUG] [1642705733.896489232] [rmw_cyclonedds_cpp]: ************ Service Details *********
[ouster_driver-1] [DEBUG] [1642705733.896500648] [rmw_cyclonedds_cpp]: Sub Topic rq/ouster_driver/get_parameter_typesRequest
[ouster_driver-1] [DEBUG] [1642705733.896503273] [rmw_cyclonedds_cpp]: Pub Topic rr/ouster_driver/get_parameter_typesReply
[ouster_driver-1] [DEBUG] [1642705733.896508065] [rmw_cyclonedds_cpp]: ***********
[ouster_driver-1] [DEBUG] [1642705733.896597982] [rcl]: Service initialized
[ouster_driver-1] [DEBUG] [1642705733.896620898] [rcl]: Initializing service for service name 'ouster_driver/set_parameters'
[ouster_driver-1] [DEBUG] [1642705733.896625065] [rcl]: Expanded service name '/ouster_driver/set_parameters'
[ouster_driver-1] [DEBUG] [1642705733.896941565] [rmw_cyclonedds_cpp]: ************ Service Details *********
[ouster_driver-1] [DEBUG] [1642705733.896952398] [rmw_cyclonedds_cpp]: Sub Topic rq/ouster_driver/set_parametersRequest
[ouster_driver-1] [DEBUG] [1642705733.896959440] [rmw_cyclonedds_cpp]: Pub Topic rr/ouster_driver/set_parametersReply
[ouster_driver-1] [DEBUG] [1642705733.896963523] [rmw_cyclonedds_cpp]: ***********
[ouster_driver-1] [DEBUG] [1642705733.897049815] [rcl]: Service initialized
[ouster_driver-1] [DEBUG] [1642705733.897065940] [rcl]: Initializing service for service name 'ouster_driver/set_parameters_atomically'
[ouster_driver-1] [DEBUG] [1642705733.897083774] [rcl]: Expanded service name '/ouster_driver/set_parameters_atomically'
[ouster_driver-1] [DEBUG] [1642705733.897801774] [rmw_cyclonedds_cpp]: ************ Service Details *********
[ouster_driver-1] [DEBUG] [1642705733.897813732] [rmw_cyclonedds_cpp]: Sub Topic rq/ouster_driver/set_parameters_atomicallyRequest
[ouster_driver-1] [DEBUG] [1642705733.897816524] [rmw_cyclonedds_cpp]: Pub Topic rr/ouster_driver/set_parameters_atomicallyReply
[ouster_driver-1] [DEBUG] [1642705733.897822107] [rmw_cyclonedds_cpp]: ***********
[ouster_driver-1] [DEBUG] [1642705733.897919857] [rcl]: Service initialized
[ouster_driver-1] [DEBUG] [1642705733.897934940] [rcl]: Initializing service for service name 'ouster_driver/describe_parameters'
[ouster_driver-1] [DEBUG] [1642705733.897941190] [rcl]: Expanded service name '/ouster_driver/describe_parameters'
[ouster_driver-1] [DEBUG] [1642705733.898335232] [rmw_cyclonedds_cpp]: ************ Service Details *********
[ouster_driver-1] [DEBUG] [1642705733.898344941] [rmw_cyclonedds_cpp]: Sub Topic rq/ouster_driver/describe_parametersRequest
[ouster_driver-1] [DEBUG] [1642705733.898347107] [rmw_cyclonedds_cpp]: Pub Topic rr/ouster_driver/describe_parametersReply
[ouster_driver-1] [DEBUG] [1642705733.898352649] [rmw_cyclonedds_cpp]: ***********
[ouster_driver-1] [DEBUG] [1642705733.898430191] [rcl]: Service initialized
[ouster_driver-1] [DEBUG] [1642705733.898467732] [rcl]: Initializing service for service name 'ouster_driver/list_parameters'
[ouster_driver-1] [DEBUG] [1642705733.898474774] [rcl]: Expanded service name '/ouster_driver/list_parameters'
[... cut ...]
[ERROR] [ouster_driver-1]: process has died [pid 221287, exit code -6, cmd '/home/bertaveira/Desktop/Parallels Shared Folders/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/ros2_ouster/ouster_driver --ros-args --log-level debug --ros-args -r __node:=ouster_driver -r __ns:=/ --params-file /home/bertaveira/Desktop/Parallels Shared Folders/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/share/ros2_ouster/params/sensor.yaml'].
[INFO] [launch.user]: [LifecycleLaunch] Ouster driver node is exiting.

```

SteveMacenski, 2022-01-20:
I'd be happy to field a PR if you find something wrong to fix, but I'm not a user of this work at the moment and don't have easy access to a sensor to debug.

bertaveira, 2022-01-20:
@SteveMacenski Thanks. I am not the most knowledgable person to solve the issue but I will try to at least provide as much information as possible. I don't have a machine with a clean galactic installed so thats why I was trying docker. All my tests were indeed with fresh docker container with only those two dependencies. 

I will try installing the dependencies via rosdep instead of manual installing those two to check if the same persists. 

In the meantime I ran it with backward_ros and got a stack trace:
```
[ouster_driver-1] ouster_driver: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:180: Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType Eigen::DenseCoeffsBase<Derived, 0>::operator()(Eigen::Index) const [with Derived = Eigen::Map<const Eigen::Array<unsigned int, -1, -1, 1, -1, -1> >; Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType = unsigned int; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.
[ouster_driver-1] Stack trace (most recent call last):
[ouster_driver-1] #19   Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/ros2_ouster/ouster_driver", at 0xaaaace8379f3, in $x
[ouster_driver-1] #18   Source "../csu/libc-start.c", line 308, in __libc_start_main [0xffff8674814f]
[ouster_driver-1] #17   Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/ros2_ouster/ouster_driver", at 0xaaaace837ffb, in main
[ouster_driver-1] #16   Object "/opt/ros/foxy/lib/librclcpp.so", at 0xffff86b91317, in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)
[ouster_driver-1] #15   Object "/opt/ros/foxy/lib/librclcpp.so", at 0xffff86b9459b, in rclcpp::executors::SingleThreadedExecutor::spin()
[ouster_driver-1] #14   Object "/opt/ros/foxy/lib/librclcpp.so", at 0xffff86b8ea13, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[ouster_driver-1] #13   Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff8704ee77, in rclcpp::GenericTimer<std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>, (void*)0>::execute_callback()
[ouster_driver-1] #12   Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff87051207, in void rclcpp::GenericTimer<std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>, (void*)0>::execute_callback_delegate<std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>, (void*)0>()
[ouster_driver-1] #11   Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff87052c87, in void std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>::operator()<, void>()
[ouster_driver-1] #10   Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff870542bf, in void std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>)
[ouster_driver-1] #9    Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff87054e8f, in std::__invoke_result<void (ros2_ouster::OusterDriver::*&)(), ros2_ouster::OusterDriver*&>::type std::__invoke<void (ros2_ouster::OusterDriver::*&)(), ros2_ouster::OusterDriver*&>(void (ros2_ouster::OusterDriver::*&)(), ros2_ouster::OusterDriver*&)
[ouster_driver-1] #8    Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff8705584b, in void std::__invoke_impl<void, void (ros2_ouster::OusterDriver::*&)(), ros2_ouster::OusterDriver*&>(std::__invoke_memfun_deref, void (ros2_ouster::OusterDriver::*&)(), ros2_ouster::OusterDriver*&)
[ouster_driver-1] #7    Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff86fc0dd3, in ros2_ouster::OusterDriver::processData()
[ouster_driver-1] #6    Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff86fc9a7f, in sensor::ScanProcessor::process(unsigned char const*, unsigned long)
[ouster_driver-1] #5    Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff86fc5aef, in ros2_ouster::toMsg(ouster::LidarScan, std::chrono::duration<long, std::ratio<1l, 1000000000l> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ouster::sensor::sensor_info const&, unsigned char, unsigned long)
[ouster_driver-1] #4    Object "/media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/libouster_driver_core.so", at 0xffff86fcd5e7, in Eigen::DenseCoeffsBase<Eigen::Map<Eigen::Array<unsigned int, -1, -1, 1, -1, -1> const, 0, Eigen::Stride<0, 0> >, 0>::operator()(long) const
[ouster_driver-1] #3    Source "/build/glibc-vWotZY/glibc-2.31/assert/assert.c", line 101, in __assert_fail [0xffff867547a3]
[ouster_driver-1] #2    Source "/build/glibc-vWotZY/glibc-2.31/assert/assert.c", line 92, in __assert_fail_base [0xffff8675473f]
[ouster_driver-1] #1    Source "/build/glibc-vWotZY/glibc-2.31/stdlib/abort.c", line 79, in abort [0xffff86747deb]
[... cut ...]
[ERROR] [ouster_driver-1]: process has died [pid 6047, exit code -6, cmd '/home/bertaveira/Desktop/Parallels Shared Folders/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/lib/ros2_ouster/ouster_driver --ros-args --log-level debug --ros-args -r __node:=ouster_driver -r __ns:=/ --params-file /home/bertaveira/Desktop/Parallels Shared Folders/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/install/ros2_ouster/share/ros2_ouster/params/sensor.yaml'].
[INFO] [launch.user]: [LifecycleLaunch] Ouster driver node is exiting.
```

SteveMacenski, 2022-01-20:
There is where compiling with debug flags would help so we could see specific lines. The `toMsg` line is this function https://github.com/ros-drivers/ros2_ouster_drivers/blob/616891c434aba58c6e1ad0a175c929fc11cf8789/ros2_ouster/include/ros2_ouster/conversions.hpp#L205-L237 which itself has no real use of Eigen. My guess is one of the `ouster::` functions does, but that's very odd if that call is what causes an exception.

bertaveira, 2022-01-20:
@SteveMacenski How would I go about using debug flags? Sorry I really never used tools like that for ros nodes

SteveMacenski, 2022-01-20:
https://navigation.ros.org/tutorials/docs/get_backtrace.html

bertaveira, 2022-01-20:
I did as it said and printed the backtrace:

```
ouster_driver: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:180: Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType Eigen::DenseCoeffsBase<Derived, 0>::operator()(Eigen::Index) const [with Derived = Eigen::Map<const Eigen::Array<unsigned int, -1, -1, 1, -1, -1> >; Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType = unsigned int; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.
--Type <RET> for more, q to quit, c to continue without paging--

Thread 1 "ouster_driver" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) 
(gdb) backtrace
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000fffff75a3dec in __GI_abort () at abort.c:79
#2  0x0000fffff75b0740 in __assert_fail_base (
    fmt=0xfffff76acc48 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0xfffff7f177c0 "index >= 0 && index < size()", 
    file=file@entry=0xfffff7f17788 "/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h", line=line@entry=180, 
    function=function@entry=0xfffff7f177e0 "Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType Eigen::DenseCoeffsBase<Derived, 0>::operator()(Eigen::Index) const [with Derived = Eigen::Map<const Eigen::Array<unsigned int, -1, -1, 1, -1, -1> >;"...) at assert.c:92
#3  0x0000fffff75b07a4 in __GI___assert_fail (
    assertion=0xfffff7f177c0 "index >= 0 && index < size()", 
    file=0xfffff7f17788 "/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h", line=180, 
    function=0xfffff7f177e0 "Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType Eigen::DenseCoeffsBase<Derived, 0>::operator()(Eigen::Index) const [with Derived = Eigen::Map<const Eigen::Array<unsigned int, -1, -1, 1, -1, -1> >;"...)
    at assert.c:101
#4  0x0000fffff7e655e8 in Eigen::DenseCoeffsBase<Eigen::Map<Eigen::Array<unsigned int, -1, -1, 1, -1, -1> const, 0, Eigen::Stride<0, 0> >, 0>::operator() (
    this=0xffffffffd558, index=-1)
    at /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:180
--Type <RET> for more, q to quit, c to continue without paging--
#5  0x0000fffff7e5daf0 in ros2_ouster::toMsg (ls=..., timestamp=..., 
    frame="laser_data_frame", mdata=..., ring_to_use=0 '\000', override_ts=0)
    at /media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:229
#6  0x0000fffff7e61a80 in sensor::ScanProcessor::process (this=0xaaaaaae0e440, 
    data=0xaaaaaadc9900 "\326\336Ey\002\002", override_ts=0)
    at /media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/ros2_ouster/include/ros2_ouster/processors/scan_processor.hpp:90
#7  0x0000fffff7e58dd4 in ros2_ouster::OusterDriver::processData (
    this=0xaaaaaab4bf80)
    at /media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/ros2_ouster/src/ouster_driver.cpp:226
#8  0x0000fffff7eed84c in std::__invoke_impl<void, void (ros2_ouster::OusterDriver::*&)(), ros2_ouster::OusterDriver*&> (
    __f=@0xaaaaaae42990: (void (ros2_ouster::OusterDriver::*)(class ros2_ouster::OusterDriver * const)) 0xfffff7e58ab4 <ros2_ouster::OusterDriver::processData()>, __t=@0xaaaaaae429a0: 0xaaaaaab4bf80) at /usr/include/c++/9/bits/invoke.h:73
#9  0x0000fffff7eece90 in std::__invoke<void (ros2_ouster::OusterDriver::*&)(), ros2_ouster::OusterDriver*&> (
    __fn=@0xaaaaaae42990: (void (ros2_ouster::OusterDriver::*)(class ros2_ouster::OusterDriver * const)) 0xfffff7e58ab4 <ros2_ouster::OusterDriver::processData()>) at /usr/include/c++/9/bits/invoke.h:95
#10 0x0000fffff7eec2c0 in std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ous--Type <RET> for more, q to quit, c to continue without paging--
ter::OusterDriver*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>) (this=0xaaaaaae42990, __args=...) at /usr/include/c++/9/functional:400
#11 0x0000fffff7eeac88 in std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>::operator()<, void>() (this=0xaaaaaae42990)
    at /usr/include/c++/9/functional:484
#12 0x0000fffff7ee9208 in rclcpp::GenericTimer<std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>, (void*)0>::execute_callback_delegate<std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>, (void*)0>() (this=0xaaaaaae42960)
    at /opt/ros/foxy/include/rclcpp/timer.hpp:223
#13 0x0000fffff7ee6e78 in rclcpp::GenericTimer<std::_Bind<void (ros2_ouster::OusterDriver::*(ros2_ouster::OusterDriver*))()>, (void*)0>::execute_callback() (
    this=0xaaaaaae42960) at /opt/ros/foxy/include/rclcpp/timer.hpp:209
#14 0x0000fffff7a26a14 in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) () from /opt/ros/foxy/lib/librclcpp.so
#15 0x0000fffff7a2c59c in rclcpp::executors::SingleThreadedExecutor::spin() ()
   from /opt/ros/foxy/lib/librclcpp.so
#16 0x0000fffff7a29318 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () from /opt/ros/foxy/lib/librclcpp.so
#17 0x0000aaaaaaab7fec in main (argc=11, argv=0xffffffffe778)
    at /media/psf/Home/Documents/CFS/dv/ROS_packages/ouster-drivers/ros2_ouster/src/main.cpp:25
(gdb)
```

bertaveira, 2022-01-20:
Ok so I checked as well changing this parameter:
https://github.com/ros-drivers/ros2_ouster_drivers/blob/e379108778c7a2ff22fdd9c04ac5170d86487ff8/ros2_ouster/params/sensor.yaml#L52
to only `IMG|PCL|IMU`

This way it would not try to generate and send the laser scan. It works without crashing. I am not totally sure why but it has something to do with the laser scan indeed.

SteveMacenski, 2022-01-20:
That helps alot - the line that the traceback gives me is then L229, which is the first `field()` call https://github.com/ros-drivers/ros2_ouster_drivers/blob/616891c434aba58c6e1ad0a175c929fc11cf8789/ros2_ouster/include/ros2_ouster/conversions.hpp#L229

Its possible the change to support FW 2.0 / 2.1 didn't properly test the laser scan section and is problematic. 

If you remove PCL and only do scan / IMU (and/or IMG) does it work? I'm curious if its less about scan and more about multiple users of the scan accumulator, which was core to the 2.0 FW change.

bertaveira, 2022-01-20:
This past tests I was running on the galactic branch (after FW 2.0 but before FW 2.1). I can confirm that no matter if I put `SCAN ` or `IMU|SCAN ` it fails. If SCAN is included it fails. Tried just now as well for the ros2 branch which is the latest and the same applies.

SteveMacenski, 2022-01-21:
OK, I can't easily test without a sensor handy but I'd like you to try the following:
- I suspect something is up with finding the ring closest to the 0-elevation marker (https://github.com/ros-drivers/ros2_ouster_drivers/blob/ros2/ros2_ouster/include/ros2_ouster/processors/scan_processor.hpp#L61-L69). This code is iterating through the N rings on your lidar and trying to find the one that is at 0-angle (planar) or as close to it as your configuration contains. Add a print here and make sure it is picking up the proper beam. Your traceback says `ring_to_use=0` which makes me think its not working, which could be causing an indexing problem
- I suspect that maybe something is wrong with the indexing on https://github.com/ros-drivers/ros2_ouster_drivers/blob/616891c434aba58c6e1ad0a175c929fc11cf8789/ros2_ouster/include/ros2_ouster/conversions.hpp#L227-L234.  The FW2.0 conversion added in that code https://github.com/ros-drivers/ros2_ouster_drivers/pull/67/files and it looks to me like that might be wrong since your traceback shows `index=-1` in the #4 eigen error call. Even if the issue in part 1 above is true, it shouldn't still cause an assertion error, it should just give you wrong data. So I think this for loop is incorrect regardless. Or for some reason the `ls.w` is `0` so the starting index is -1 due to some missing metadata being populated for this processor type.

bertaveira, 2022-01-21:
1. It makes sense given that the Lidar is the Bellow Horizon configuration. This is the beam are aligned from 0º to -22.5º linearly distributed. Here are the prints of the variables:
```
[ouster_driver-1] _ring = 0
[ouster_driver-1] zero_angle = 0.35
```
Working on the second point you brought up as we speak

bertaveira, 2022-01-21:
Wow you hit the jackpot. It is crashing inside the loop on the first push (`msg.ranges.push_back(...)`) . I verified this by printing before and after the function call as well as after the loop and it seems to crash in that one. I added prints of the variables `ring_to_use`, `ls.w` as well as the variable `i` in all loop iterations. Here is a screenshot of the code with the prints:

[image]

As you suspected the variable `i` is reaching a very weird value. Here is the output to the terminal (removed a bit):

```
[ouster_driver-1] ring_to_use = 0
[ouster_driver-1] ls.w = 512
[ouster_driver-1] i = 511
[ouster_driver-1] i = 510
[ouster_driver-1] i = 509
[ouster_driver-1] i = 508
[ouster_driver-1] i = 507
(...) Removed since it was basically just a count down
[ouster_driver-1] i = 7
[ouster_driver-1] i = 6
[ouster_driver-1] i = 5
[ouster_driver-1] i = 4
[ouster_driver-1] i = 3
[ouster_driver-1] i = 2
[ouster_driver-1] i = 1
[ouster_driver-1] i = 0
[ouster_driver-1] i = 18446744073709551615
```

bertaveira, 2022-01-21:
It makes sense to crash. It is a loop with `i >= 0` but `i` can only be positive.

Changing the line to:
```
for (size_t i = ls.w * ring_to_use + ls.w ; i > ls.w * ring_to_use; i--) {
```
Fixes the crash but not sure if it does exactly the same

SteveMacenski, 2022-01-21:
`18446744073709551615 == -1` (2^64) for 64 bit systems. So its indexing incorrectly and getting down to -1. I think then you just need a quick update to that for loop to correct the iteration bounds. 

What's the iteration range of your change? should be 0 to resolution of lidar in ticks - 1. (e.g. 0-511 for a 512 setting). 

That's probably the fix right there that you list, just verify quickly the range is correct. If you submit a PR, I can merge, backport, and re-release

SteveMacenski, 2022-01-21:
> Wow you hit the jackpot

That's why I asked for the backtrace :wink: these 2 lines gave me all the context clues to narrow into those 2 parts of the code

```
#4  0x0000fffff7e655e8 in Eigen::DenseCoeffsBase<Eigen::Map<Eigen::Array<unsigned int, -1, -1, 1, -1, -1> const, 0, Eigen::Stride<0, 0> >, 0>::operator() (
    this=0xffffffffd558, index=-1)
    at /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:180
--Type <RET> for more, q to quit, c to continue without paging--
#5  0x0000fffff7e5daf0 in ros2_ouster::toMsg (ls=..., timestamp=..., 
    frame="laser_data_frame", mdata=..., ring_to_use=0 '\000', override_ts=0)
```

bertaveira, 2022-01-21:
> That's probably the fix right there that you list, just verify quickly the range is correct. If you submit a PR, I can merge, backport, and re-release

Which branch do you want me to change? ros2?

SteveMacenski, 2022-01-21:
yes please

bertaveira, 2022-01-22:
Well the for loop is actually correct in the index the problem is that we are subtracting to an unsigned type of variable which results in it being impossible to be negative. Since we say that it should run for any `i >= 0` it will subtract resulting in overflow to the highest possible positive number and continue running since the condition still meets.

Fun bug to find. Had to actually try it to verify it actually happens on for loops of unsigned types: https://onlinegdb.com/EinwrMY2u

I can see multiple fixes. The two that sound the cleanest are: 
- Adding `if (i == 0) break;` ate the end of the for loop
- Replacing the for loop with a `do { } while(i > 0...)`

PR #92 (2022-01-27): fix crash indexing for laser scan creation
Closes #90
closed 2022-01-27 by SteveMacenski