ros-drivers/ros2_ouster_drivers #91: Does the driver publish orientation of the IMU?

Sollimann, 2022-01-21:
I'm wondering if the ros2 driver produces orientation of the IMU on the `/imu` topic, or do I need to run the IMU-topic through a complementary filter lick Madgwick to produce the orientation?

I found this old thread [here](https://github.com/ouster-lidar/ouster_example/issues/307) saying here that the ouster-lidar does not provide orientation, but it might be different for this particular ouster driver?

Thanks!

bertaveira, 2022-01-21:
It does send the message but like the thread says it does not provide value since the IMU does not estimate it. I can confirm that angular velocity and acceleration work but the velocity is quite bad (probably due to not having a GPS connected to it).

You can see what the IMU sends in this screenshot:
[image]
All orientation messages are either 0 or 1 constant.

Sollimann, 2022-01-21:
Thanks for the answer!

closed 2022-01-21 by SteveMacenski