ros-drivers/ros2_ouster_drivers #102: Launching Multiple Ousters

Zaaler, 2022-05-11:
Hi, 

I have a system using two ouster lidars. I can successfully launch one lidar but, the second lidar fails to launch.

Has anyone had luck launching two lidars?

I added params_path and node_name launch variables that I am happy to submit a PR on for the open source world. So, the nodes have unique names ouster_driver_1 and ouster_driver_2 and unique parameter files. The problem is when it trys to open a connection to the host computer I believe.

Any help is appreciated. And thank you very much for your work on these drivers.

Z

SteveMacenski, 2022-05-11:
Did you try to set the distinct IP addresses for the lidars to connect to for each node? There's a parameter for the IP to use that can be different for both nodes. From glancing at the SDK, it seems like multiple connections shouldn't be a problem if you explicitly state the lidar IP for each node to use. Nothing about ethernet connections should block it, but I also have not tried this myself, I don't actually have more than 1 unit.

Zaaler, 2022-05-11:
Yeah I have them set to unique IP addresses. on my system they ar 192.xx.xx.45 and 46. Continuing to work on it today. I'll update soon.

Zaaler, 2022-05-11:
This is the error message;
```
May 11 08:07:40 AIONio-Flash ouster_start.sh[12620]: [ouster_driver-1] [INFO] [1652278060.690230100] [ouster_driver_1]: Configuring Ouster driver node.
May 11 08:07:40 AIONio-Flash ouster_start.sh[12620]: [ouster_driver-1] [INFO] [1652278060.690457967] [ouster_driver_1]: Connecting to sensor at 192.xx.xx.45.
May 11 08:07:40 AIONio-Flash ouster_start.sh[12620]: [ouster_driver-1] [INFO] [1652278060.690516413] [ouster_driver_1]: Sending data from sensor to 192.xx.xx.128.
May 11 08:07:40 AIONio-Flash ouster_start.sh[12620]: [ouster_driver-1] udp bind(): Transport endpoint is not connected
May 11 08:07:40 AIONio-Flash ouster_start.sh[12620]: [ouster_driver-1] udp bind(): Transport endpoint is not connected
May 11 08:07:40 AIONio-Flash ouster_start.sh[12620]: [ouster_driver-1] [FATAL] [1652278060.690941275] [ouster_driver_1]: Exception thrown: (Failed to create connection to lidar.)
May 11 08:07:40 AIONio-Flash ouster_start.sh[12620]: [ERROR] [ouster_driver-1]: process has died [pid 12979, exit code 255, cmd '/home/aion/aion_ws/install/ros2_ouster/lib/ros2_ouster/ouster_driver
```

Zaaler, 2022-05-11:
UPDATE: I was able to get them both to launch successfully by changing the UDP Port of my second lidar to 7505. But, I'm not getting the points to come through yet. Any suggestions much appreciated. I'm not super knowledgeable in networking so, there probably is a better way to do this.

Zaaler, 2022-05-11:
UPDATE: I was able to successfully get both lidars running and reading. I had to make a change that allowed me to specify the remapped topics based on the node name. I have the code that allows you to specify a params file and the node name and remap topics to provide unique topic names based on the node name. I am willing to contribute these changes if you are interested in adding that functionality. Again, thank you for your response @SteveMacenski and for the work you have put into this driver.

Z

closed 2022-05-11 by Zaaler

szepilot, 2023-01-23:
> UPDATE: I was able to successfully get both lidars running and reading. I had to make a change that allowed me to specify the remapped topics based on the node name. I have the code that allows you [...]

hi @Zaaler Can you share the code? We are working on the same problem. Whenever we are modifying the namespace or remapping in the launch it falls back to the default parameters for some reason. Thanks

szepilot, 2023-01-23:
We also managed to solve, so i just link it here: https://github.com/jkk-research/lexus_bringup/tree/main/launch/drivers

nsteve2407, 2023-02-24:
Had the same issue. To fix this each osuter driver node should:
1. Have different namespace
2. Different node name
3. Different udp ports