ros-drivers/ros2_ouster_drivers #105: Support Low Data Rate Profile

vinnnyr, 2022-06-20:
Firmware 2.3 allowed for the ability to request different data packet formats from the sensor. [1]. The Low Data Rate profile might help with robots with limited compute / network bandwidth [2].

Looks like the `ouster_example` repo added a configurable UDPProfile for this [3]

1. https://ouster.com/blog/firmware-2-3-lower-data-rates-and-new-sensor-telemetry/
2. Section 6 (page 26): https://data.ouster.io/downloads/software-user-manual/firmware-user-manual-v2.3.0.pdf
3. https://github.com/ouster-lidar/ouster_example/blob/64a6cbe6dff1eea0cb5a79d63b79acf11b68b16d/ouster_client/include/ouster/types.h#L161

Where is the best place to implement this for this driver?

vinnnyr, 2022-06-20:
I think the driver would already handle getting the correct packet format, since it seems like everything is based on the underlying Ouster library `getPacketFormat`

https://github.com/ros-drivers/ros2_ouster_drivers/blob/2d36cbd7ceac9dfa1bea206439a10d4b0ca731cb/ros2_ouster/src/sensor.cpp#L157

I would have to set the profile manually, and see if that is right :thinking:

vinnnyr, 2022-07-21:
on a similar note, when trying to apply dual return packet mode, it seems like this breaks the driver. I get the following log spammed:
```
[ouster_driver-1] Unexpected udp packet length: 6465
```

SteveMacenski, 2022-08-17:
If its a direct copy over from `ouster_example` we can do that, that's basically what we do right now because the ouster SDK is source controlled with the ROS 1 drivers and not actually released into a binary format we can use here to share between them (hence, the unfortunate copy paste)

Grisly00, 2022-09-28:
Hi! 
Is there any plans on when this feature will be available? We would really like to test the dual return mode of the new lidars. 
thx

vinnnyr, 2022-09-28:
i'm not actively working on this at the moment -- i'm not sure if anyone else is.

vinnnyr, 2022-10-14:
@SteveMacenski  can you explain a bit what the process is to copy over from the `ouster_example` repo? I might just be missing something, but I don't really see a clear copy paste from the other repo.

SteveMacenski, 2022-10-31:
A matched set of header files that works for a paired version of the Ouster lidar firmware versions. Since Ouster isn't careful about backward compatibility, once we release a version of this software to a ROS distribution, it has to continue to work for whatever firmware versions for the life of that distribution. As a result, you cannot upgrade your sensor's firmware and use this driver on a given ROS 2 distribution unless its backward compatible to work with that same version of the ouster sdk given via ouster_example.