ros-drivers/ros2_ouster_drivers #109: ParameterTypeException

carmiac, 2022-08-16:
I recently updated from foxy to humble and now my ouster nodes crash with the following message:
```
[ouster_driver-1] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
[ouster_driver-1]   what():  expected [string] got [not set]

```
I've tried using the ros-humble-ros2-ouster package as well as building from both the ros2 and the humble branch all with the same issue.  My params.yaml file is:
```
lidar_1:
  ouster_537:
    ros__parameters:
      lidar_ip: os-992129000537.local
      computer_ip: 192.168.42.2
      lidar_mode: "1024x10"
      imu_port: 7503
      lidar_port: 7502
      sensor_frame: lidar_sensor_frame
      laser_frame: lidar_data_frame
      imu_frame: lidar_imu_frame
      use_system_default_qos: True
      timestamp_mode: TIME_FROM_ROS_RECEPTION
      proc_mask: PCL
```

Previously this was working, both in this small demo and in a much larger system.

SteveMacenski, 2022-08-17:
Are you sure that your parameter file namespaces are correct? That file looks right to me after the `ros__parameters` portion, but it could be that your node isn't named `lidar_1.ouster_537` as its trying to find the parameters under `lidar_1.ouster_537.<params>`.

carmiac, 2022-08-22:
That was it, missed it in a refactor.

closed 2022-08-22 by carmiac