ros-drivers/ros2_ouster_drivers #111: Problem with ros2_ouster package using colcon build

orjano-max, 2022-09-12:
Have a problem building the ros2_ouster package with colcon build.

using ros2 foxy on ubuntu 20.04
cloned using: git clone -b foxy-devel https://github.com/ros-drivers/ros2_ouster_drivers
dependencies: rosdep install --from-paths src --ignore-src --rosdistro=foxy -y
then build: colcon build

Get sthis output:

Starting >>> ouster_msgs
Finished <<< ouster_msgs [0.49s]                     
Starting >>> ros2_ouster
--- stderr: ros2_ouster                         
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:62 (message):
  common is required but boost was not found
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:367 (pcl_report_not_found)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  CMakeLists.txt:17 (find_package)

---
Failed   <<< ros2_ouster [1.32s, exited with code 1]

Summary: 1 package finished [2.00s]
  1 package failed: ros2_ouster
  1 package had stderr output: ros2_ouster

SteveMacenski, 2022-09-12:
You’re missing and install of PCL.

closed 2022-09-12 by SteveMacenski

TZECHIN6, 2022-10-03:
I have the same issue, however I do install PCL by below commands:
```
$ sudo apt install libpcl-dev
$ sudo apt isntall libpcl-conversions-dev
```
So which PCL do I miss? So that I can install it back. Thx