ros-drivers/ros2_ouster_drivers #116: Does not work with sensor on firmware v2.4

vinnnyr, 2022-10-14:
Details / logs to come. Just a warning I wanted to post right now to users of the repo, to not upgrade until this is resolved, since it is not possible to revert to firmware v2.3 (they disabled the download on the website).

From what I understand, this is due to some deprecated parameters in the new firmware (`auto_start_flag` and `udp_ip`). 

I am working on a fix, and will try to PR it when I have tested it against the new firmware, however I am not really sure how backwards compatible it will be.

vinnnyr, 2022-10-19:
From Ouster, this is the patch that would work for 2.4 on their version of the driver.
https://github.com/ouster-lidar/ouster_example/commit/6e3df5efcb332562c6381e5b668c6936c09d646a

@SteveMacenski  would you be ok with a PR back here with the content of that diff? The only thing is that pushing for firmware 2.4 compatiblity might break compatible with earlier firmwares

kpsgf7, 2022-10-19:
@vinnnyr Are you attempting a fix for this then? I may attempt a fix if we are unable to get v2.3 of the firmware from Ouster. Would love to contribute if that's something that would be helpful here.

vinnnyr, 2022-10-19:
@kpsgf7  can you test this fork? https://github.com/fireflyautomatix/ros2_ouster_drivers/tree/fix-for-firmware-2.4

I currently don't have a lidar with v2.4 to test on available to me, hence my hesitation in opening the PR

vinnnyr, 2022-10-20:
I just tested this with 2 v2.4 lidars. I think that PR should be ready for review now!

kpsgf7, 2022-10-20:
This worked for us! Thank you! We really appreciate it.

closed 2022-11-28 by SteveMacenski