ros-drivers/ros2_ouster_drivers #117: [feature request] Control Azimuthal Window with Driver

vinnnyr, 2022-10-14:
It would be nice to be able to control the azimuthal window of the lidar via a parameter file rather than control it using the firmware page
[image]

SteveMacenski, 2022-10-14:
If there's a parameter in the SDK that lets us control that, we can certainly expose it! I see no qualms with that.

vinnnyr, 2022-10-14:
after digging I realized someone else proposed this: https://github.com/ros-drivers/ros2_ouster_drivers/pull/98 and the underlying version here: https://github.com/ouster-lidar/ouster_example/pull/366

bertaveira, 2022-11-05:
The problem was that after discussing in the main repo it was mentioned that azimuth window change in the lidar config requires a restart of the lidar (not manually but it has to internally restart). Therefore having that as a parameter would make the lidar have to start up twice taking twice as long to start then.

SteveMacenski, 2022-11-05:
I can see their worry, but personally I'd say "OK" to that and just make sure its both documented and when its changed a warning is thrown to let someone know. If startup time is something they care about, then they can make that change more permanently elsewhere to help here - but that way they know but isn't blocking others.