ros-drivers/ros2_ouster_drivers #119: Failed to create connection to lidar.

stapperfend, 2022-11-19:
Hello,
I just received the OS0-128 Rev6 and want to visualize the sensor data.
I have the firmware v2.4.0 installed and connected the sensor via ethernet to an NVIDIA Jetson AGX Orin, running Ubuntu 20.04 with Foxy.

I cloned this git, compiled everything with colcon and followed the IPv4 setup provided in this git. I am using the eth0 connection - the network is UP.

I changed in the driver_config.yaml:
`lidar_ip: 10.5.5.84      
computer_ip: 10.5.5.1   `

When I launch 
`ros2 launch ros2_ouster driver_launch.py`
I get the following output:

> ros2 launch ros2_ouster driver_launch.py
> [INFO] [launch]: All log files can be found below /home/orinone/.ros/log/2022-11-19-14-12-25-763577-orinone-3976
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [ouster_driver-1]: process started with pid [3988]
> [ouster_driver-1] [INFO] [1668863546.184379897] [ouster_driver]: Configuring Ouster driver node.
> [ouster_driver-1] [INFO] [1668863546.184656701] [ouster_driver]: Connecting to sensor at 10.5.5.84.
> [ouster_driver-1] [INFO] [1668863546.184684766] [ouster_driver]: Sending data from sensor to 10.5.5.1.
> [ouster_driver-1] [FATAL] [1668863577.584329225] [ouster_driver]: Exception thrown: (Failed to create connection to lidar.)
> [ERROR] [ouster_driver-1]: process has died [pid 3988, exit code 255, cmd '/home/orinone/ros2_ws/install/ros2_ouster/lib/ros2_ouster/ouster_driver --ros-args --log-level INFO --ros-args -r __node:=ouster_driver -r __ns:=/ --params-file /home/orinone/ros2_ws/install/ros2_ouster/share/ros2_ouster/params/driver_config.yaml'].

The LIDAR starts spinning during this procedure.  

Any idea why no connection to the lidar can be established?

vinnnyr, 2022-11-22:
Please try this branch:
https://github.com/ros-drivers/ros2_ouster_drivers/pull/118

firmware v2.4 removed deprecated parameters that the driver off of `main` currently does not account for. 

On that note, I need to prioritize getting that branch merged.

stapperfend, 2022-11-25:
The #118 branch doesn't work for us. There is still no connection possible.
When we start the ros2 launch command with the IPv4 setup, the LIDAR starts spinning and also goes into the "RUNNING" State. Nonetheless, we still get the same error.
When we read the transition topic in ROS2, we can also see the planned transition from state 1 to 10 (not 100% sure about these numbers), but then the error comes again.

vinnnyr, 2022-11-26:
Can you share some of the logs from this issue? Curious that the branch did not resolve the issue -- you are the first report that hasn't had luck and the "failed to connect to lidar" issue with firmware 2.4 is exactly what I have seen prior to that branch.

Are you doing any other configuration changes? Are you comfortable sharing your parameter file?

stapperfend, 2022-11-28:
After a new setup of the OS, it seems to run now with the #118 firmware fix. Not exactly sure why. 

We now get the following terminal output:

> 
> orintwo@orintwo:~/ros2_ws$ ros2 launch ros2_ouster driver_launch.py
> [INFO] [launch]: All log files can be found below /home/orintwo/.ros/log/2022-11-28-09-53-55-469479-orintwo-6854
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [ouster_driver-1]: process started with pid [6866]
> [ouster_driver-1] [INFO] [1669625635.898379964] [ouster_driver]: Configuring Ouster driver node.
> [ouster_driver-1] [INFO] [1669625635.898749237] [ouster_driver]: Connecting to sensor at 10.5.5.84.
> [ouster_driver-1] [INFO] [1669625635.898788405] [ouster_driver]: Sending data from sensor to 10.5.5.1.
> [ouster_driver-1] [INFO] [1669625671.064975066] [ouster_driver]: This driver is compatible with sensors running fw 2.x.
> [INFO] [launch.user]: [LifecycleLaunch] Ouster driver node is activating.
> [ouster_driver-1] [INFO] [1669625671.112140435] [ouster_driver]: Activating Ouster driver node.

Is this the desired final output?  We are wondering, because the LIDAR starts spinning for around 90 seconds after it moved from state 1 -> 10 -> 2 -> 13 -> 3 (active) and then stops. We can read the topics from the LIDAR during this time. After it stops, no new messages are published.

Another issue I faced: When I want to display the PointCloud2 data (With Reliability: Best Effort and changed Frame) in RVIZ2, it says 

> Showing [131072] points from [1] message
> 550 messages received

but it only displays 1 point in RVIZ2. I checked the /points messages and the "point_step" is at 48 and the "row_step" is always at 49152 in all messages. But I see, that some of the "data" field is changing with `ros2 topic echo /points`.

SteveMacenski, 2022-11-28:
> v2.4.0 installed and connected the sensor via ethernet to an NVIDIA Jetson AGX Orin, running Ubuntu 20.04 with Foxy.

This is not a supported configuration. You can see which firmware versions each distribution supports. Because Ouster doesn't version control / make their firmware backward compatible very well, each ROS distribution has to use a static range of firmwares so we can release stable binaries for the life-cycle of the distribution. 

Downgrade to 1.x or upgrade ROS distribution / use the newer branches for v2.4 support (rolling as merged right now would support).

closed 2022-11-28 by SteveMacenski