ros-drivers/ros2_ouster_drivers #120: [feature erquest] Syncronization with camera

adrian-soch, 2022-12-02:
I'm looking for the simplest solution to sync a ROS2 lidar node with a camera node. Both camera and lidar drivers are capable of running at 10Hz, however I want to ensure that they capture data at approximately the same time. The Ouster doc has a section on [Multi-Sensor Synchronization](https://static.ouster.dev/sensor-docs/image_route1/image_route2/time_sync/time-sync.html). 

There seems to be 2 options. Using the [Phase lock](https://static.ouster.dev/sensor-docs/image_route1/image_route2/time_sync/time-sync.html#phase-lock) feature, where the external source syncs both the camera and lidar. They say `Phase locking is useful for synchronizing a sensor with other devices including camera` but I'm not sure how to implement with the ROS2 driver setup. Another option is the lidar [sends a pulse](https://static.ouster.dev/sensor-docs/image_route1/image_route2/time_sync/time-sync.html#external-trigger-clock-source) to trigger the camera.

Any insight to this problem would be appreciated.

SteveMacenski, 2022-12-02:
You'd have to set that up on your sensor itself.

adrian-soch, 2023-01-08:
Decided to add a data processor for camera image and just built a driver that combines the LiDAR and camera in one. It is the easiest solution for cameras without hardware triggers and if you are using a NIC without hardware timers.

closed 2023-01-08 by adrian-soch