ros-drivers/ros2_ouster_drivers #121: Node seg faults with 32 channel OS1 but not with 64 channel

kagibson, 2022-12-06:
Hi there,

I have two OS1's connected to an arm64-based controller running Foxy. One is a 64 channel and the other is a 32 channel. When I launch the node for the 64 channel, it works and I get a point cloud. However, when I launch the 32 channel, I get a seg fault in the node. 
I added backwards_ros to the node to get the stack trace and I've attached it. I've also attached the parameter files for both nodes.
[ouster-segfault-stacktrace.txt](https://github.com/ros-drivers/ros2_ouster_drivers/files/10159503/o
[os1_ch64_params.txt](https://github.com/ros-drivers/ros2_ouster_drivers/files/10159514/os1_ch64_params.txt)
uster-segfault-stacktrace.txt)
[ouster-curl-data.txt](https://github.com/ros-drivers/ros2_ouster_drivers/files/10159552/ouster-curl-data.txt)

[os1_ch32_params.txt](https://github.com/ros-drivers/ros2_ouster_drivers/files/10159509/os1_ch32_params.txt)

vinnnyr, 2022-12-08:
I think to unblock you you can change this line in the params:  `os1_proc_mask: IMG|PCL|IMU|SCAN` 

From the trace  you can see it's coming from the scan processor
`[ouster_driver-1] #0  | Source "/workspace/src/ouster_driver/include/ros2_ouster/processors/scan_processor.hpp", line 80, in toMsg`

Try setting this parameter to the following if you don't need the SCAN functionality.
`os1_proc_mask: IMG|PCL|IMU`

See if that helps and please report back. Then we can focus on how to fix the segfault for SCAN.

kagibson, 2022-12-16:
We tried removing the SCAN functionality and the node doesn't crash anymore, however, we don't get a point cloud. If we connect to the LiDAR with a laptop and Ouster Studio we can see a point cloud there. We'll try sniffing the packets with Wireshark and see what the LiDAR is sending out when the node is up.