ros-drivers/ros2_ouster_drivers #128: [ROS2 Foxy] Unable to build from master

brendoncintas, 2023-02-02:
Hello, 

I've got an OS1 running firmware 2.4 that's connected to an Nvidia Jetson Xavier NX. I'm attempting to build from source cloned off of the master branch, but every time I build, I resulted in the following error: 

`Starting >>> ouster_msgs
Finished <<< ouster_msgs [2.34s]                     
Starting >>> ros2_ouster
--- stderr: ros2_ouster                         
CMake Error at /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:62 (message):
  common is required but boost was not found
Call Stack (most recent call first):
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:367 (pcl_report_not_found)
  /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  CMakeLists.txt:23 (find_package)

---
Failed   <<< ros2_ouster [5.93s, exited with code 1]
                                
Summary: 1 package finished [8.93s]
  1 package failed: ros2_ouster
  1 package had stderr output: ros2_ouster
`

I followed the same procedures on another computer, and the build was successful, which leads me to believe that I do have a dependency error. After confirming with `rosdep install --from-paths src` within the ros2_ouster directory, I still result in the same error. After searching the issues within this repository, I referenced #111 thinking I'm missing the PointCloud Library (PCL), but it seems the issue goes deeper. I wonder if there's an issue with the architecture of the Jetson - which uses ARM64 as opposed to x86_64. 

Would I be able to receive guidance on if there's something I can do the CMakeLists.txt file to support ARM64?

SteveMacenski, 2023-02-02:
It may be possible some of your dependencies are not binaries on ARM or arent working with the right versions

brendoncintas, 2023-02-02:
Steve, 

I'm grateful for your lightning-fast response. I've figured that could also be the case. NVIDIA Jetsons have a tendency to have proprietary versions of the same libraries that come baked into their images, so I'm intending on trying to remake a new image for my Jetson and try recompiling from source. 

For reference, is there a list of dependencies I can look at to make sure I've got all my ducks in a row?

SteveMacenski, 2023-02-02:
The package.xml file would be it! You can cross reference any non-ros-named keys in the ros/rosdistro repo for your particular distribution for urls to source (if nonobvious)

brendoncintas, 2023-02-02:
Sweet! I'll let you know of any updates. Hopefully, it'll be one that allows me to close this ticket. 👍

brendoncintas, 2023-02-02:
Hi there, 

So I flashed the Jetson once again and pulled from master once again. Now, it looks like it can detect `boost` within the common for PCLConfig; however, upon a new build attempt, I result with a new error, shown below:

```
Starting >>> ouster_msgs
Finished <<< ouster_msgs [2.53s]                     
Starting >>> ros2_ouster
[Processing: ros2_ouster]                             
--- stderr: ros2_ouster                               
In file included from /home/jackal/ros2_ouster_drivers/ros2_ouster/src/sensor_tins.cpp:19:
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/sensor_tins.hpp:18:10: fatal error: tins/packet.h: No such file or directory
   18 | #include <tins/packet.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ouster_driver_core.dir/build.make:115: CMakeFiles/ouster_driver_core.dir/src/sensor_tins.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/jackal/ros2_ouster_drivers/ros2_ouster/src/driver_types.cpp:20:
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/sensor_tins.hpp:18:10: fatal error: tins/packet.h: No such file or directory
   18 | #include <tins/packet.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ouster_driver_core.dir/build.make:63: CMakeFiles/ouster_driver_core.dir/src/driver_types.cpp.o] Error 1
In file included from /home/jackal/ros2_ouster_drivers/ros2_ouster/src/ouster_driver.cpp:28:
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/sensor_tins.hpp:18:10: fatal error: tins/packet.h: No such file or directory
   18 | #include <tins/packet.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ouster_driver_core.dir/build.make:76: CMakeFiles/ouster_driver_core.dir/src/ouster_driver.cpp.o] Error 1
In file included from /home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/processors/image_processor.hpp:26,
                 from /home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/processors/processor_factories.hpp:23,
                 from /home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/sensor.hpp:21,
                 from /home/jackal/ros2_ouster_drivers/ros2_ouster/src/sensor.cpp:20:
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:194:26: error: ‘PointCloud2’ in namespace ‘sensor_msgs::msg’ does not name a type
  194 | inline sensor_msgs::msg::PointCloud2 toMsg(
      |                          ^~~~~~~~~~~
In file included from /home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/processors/processor_factories.hpp:25,
                 from /home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/sensor.hpp:21,
                 from /home/jackal/ros2_ouster_drivers/ros2_ouster/src/sensor.cpp:20:
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/processors/pointcloud_processor.hpp: In member function ‘virtual bool sensor::PointcloudProcessor::process(const uint8_t*, uint64_t)’:
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/processors/pointcloud_processor.hpp:117:28: error: no matching function for call to ‘toMsg(pcl::PointCloud<ouster_ros::Point>&, std::chrono::nanoseconds, std::string&, const uint64_t&)’
  117 |         _frame, override_ts));
      |                            ^
In file included from /home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/processors/image_processor.hpp:26,
                 from /home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/processors/processor_factories.hpp:23,
                 from /home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/sensor.hpp:21,
                 from /home/jackal/ros2_ouster_drivers/ros2_ouster/src/sensor.cpp:20:
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:84:35: note: candidate: ‘ouster_msgs::msg::Metadata ros2_ouster::toMsg(const ros2_ouster::Metadata&)’
   84 | inline ouster_msgs::msg::Metadata toMsg(const ros2_ouster::Metadata & mdata)
      |                                   ^~~~~
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:84:35: note:   candidate expects 1 argument, 4 provided
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:104:45: note: candidate: ‘geometry_msgs::msg::TransformStamped ros2_ouster::toMsg(const Eigen::Matrix<double, 4, 4, 2>&, const string&, const string&, const rclcpp::Time&)’
  104 | inline geometry_msgs::msg::TransformStamped toMsg(
      |                                             ^~~~~
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:105:57: note:   no known conversion for argument 1 from ‘pcl::PointCloud<ouster_ros::Point>’ to ‘const Eigen::Matrix<double, 4, 4, 2>&’
  105 |   const Eigen::Matrix<double, 4, 4, Eigen::DontAlign> & mat, const std::string & frame,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/jackal/ros2_ouster_drivers/ros2_ouster/include/ros2_ouster/conversions.hpp:135:30: note: candidate: ‘sensor_msgs::msg::Imu ros2_ouster::toMsg(const uint8_t*, const string&, const ouster::sensor::packet_format&, uint64_t)’
  135 | inline sensor_msgs::msg::Imu toMsg(
      |                              ^~~~~
[... cut ...]
/usr/lib/aarch64-linux-gnu/libpcl_octree.so.1.10
/usr/lib/aarch64-linux-gnu/libpcl_sample_consensus.so
/usr/lib/aarch64-linux-gnu/libpcl_features.so.1.10
/usr/lib/aarch64-linux-gnu/libpcl_registration.so.1.10
/usr/lib/aarch64-linux-gnu/libpcl_common.so.1.10.0
/usr/lib/aarch64-linux-gnu/libpcl_search.so.1.10
/usr/lib/aarch64-linux-gnu/libpcl_tracking.so
```
Any help on this would be greatly appreciated - my brain has been wracked all night on this now with the PCLConfig error no longer being present.

brendoncintas, 2023-02-02:
**It's fixed!**

Apparently, `rosdep install` doesn't seem to pick up ros-<version>-pcl-ros and ros-version-<pcl>-conversions as requisite dependencies to build the ros2_ouster package, resulting in this error. Installing them respectively resolved the problem, and I now have a completely built package. 

I'll be closing this issue now. Thanks!

closed 2023-02-02 by brendoncintas

brendoncintas, 2023-02-08:
Briefly opening this because I found a surefire way of resolving the error shown about PCLConfig. 

When rebuilding after a config change, I ran into the error again. I went through the PCLConfig.cmake file because now, at this point, the error falls entirely on PCLConfig being unable to find boost. 

In the cmake file, there's a set() for finding boost with the environment variable NO_BOOST_CMAKE. Here's the snippet:

```
# Disable the config mode of find_package(Boost)
  set(Boost_NO_BOOST_CMAKE ON)
  find_package(Boost 1.55.0 ${QUIET_} COMPONENTS system filesystem date_time iostreams)
```

My solution was to just outright remove that set(), and disable the ${QUIET_} within find_package. Suddenly, it worked. Here's the code I replaced:

```
  # Disable the config mode of find_package(Boost)
  find_package(Boost 1.55.0 COMPONENTS system filesystem date_time iostreams)
```

Addmittedly, not sure why this works, but as they say in software engineering: if it works, don't question it. In case someone else has this problem with their Jetson, they should try this and hopefully it'll work.