ros-drivers/ros2_ouster_drivers #129: Problem with connecting to Ouster OS0 on ros:rolling

hodnajit, 2023-02-07:
Hi,
I am facing a strange issue. I am working on Ubuntu 22.04 with ros2 rolling, and I cannot connect to the ouster sensor with your driver if I use the APT package of ros2_ouster.

My lidar is os0 32 rev 6.

My version of APT packages is
`ros-rolling-ros2-ouster is already the newest version (0.4.0-1jammy.20221229.231648)`

The connection and data stream works with Ouster Studio
[image]

But when I try to launch the driver
`ros2 launch ros2_ouster driver_launch.py params_file:="/home/hodnajit/rtabmap_experiments/install/optimus_bringup/share/optimus_bringup/params/ouster_params.yaml"`

I got 
`[ouster_driver-1] [FATAL] [1675777304.774110148] [ouster_driver]: Exception thrown: (Failed to create a connection to lidar.)`

[image]

If I run the driver from the source (branch ros2), everything works fine. Is there a bug in the current APT package?

Thanks for your work!

Best regards

Jitka

SteveMacenski, 2023-02-07:
https://github.com/ros-drivers/ros2_ouster_drivers/commits/ros2 The last 6 commits you see there are the difference between the release binaries and source code. 

It looks like you have firmware 2.4 which needs `d59df70a6e58e646fd305cb913fa8c802fc6588c`. I can run a sync though to Rolling so the binaries work on Friday when I do my master syncs across all the repos I maintain

hodnajit, 2023-02-07:
Thanks for reply :) If it is ok to release it in next rolling release, it would be great. :) 

I am ok with the answer, I was afraid, that I am doing something wrong and I wanted to report it. :) I close this issue. Thanks

closed 2023-02-07 by hodnajit