ros-drivers/ros2_ouster_drivers #131: Any plans with this ros2_ouster_driver if there is published the official one?

hodnajit, 2023-02-23:
Hi,
I am working with your ros2_ouster_drivers currently, but I have noticed, that the ouster company released their own driver
https://github.com/ouster-lidar/ouster-ros/tree/ros2

Are there any plans with this driver? I know, that currently there are APT packages for ros:rolling according to this driver, but the official driver has the same package ouster_msgs. I mean, the naming is the same, the function not. Do you have any ideas?

Thanks

SteveMacenski, 2023-02-23:
This is a good question - they haven't been in contact with me regarding it so I can't say much on the topic. When they want to release the code, they're going to run into some issues with the collisions. If they want to support it themselves, I have no issue allowing them to do so, but we probably need to coordinate a bit. Maybe now if they have to manage their own ROS drivers they'll understand how important it is to match up distributions with firmware/SDK versions :wink:

hodnajit, 2023-02-27:
Thanks for quick response :)

closed 2023-02-27 by hodnajit