ros-drivers/ros2_ouster_drivers #136: [bug]Too heavy bandwidth when multiple LiDAR plugged in

ahuazuipiaoliang, 2023-05-10:
# Too heavy bandwidth when multiple LiDAR plugged in
## System Information
ROS: Humble
LiDAR: 
OS1-32
[image]
Driver Version: https://github.com/ros-drivers/ros2_ouster_drivers/commit/5425ed023d3c1c5293a9af90d2ae81c5df889334

## Description
### launch file
```python3
#!/usr/bin/python3
from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch_ros.actions import LifecycleNode
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch.actions import EmitEvent
from launch.actions import RegisterEventHandler
from launch_ros.events.lifecycle import ChangeState
from launch_ros.events.lifecycle import matches_node_name
from launch_ros.event_handlers import OnStateTransition
from launch.actions import LogInfo
from launch.events import matches_action
from launch.event_handlers.on_shutdown import OnShutdown

import lifecycle_msgs.msg
import os

def generate_launch_description():
    share_dir = get_package_share_directory('robobus_sensor_kit_launch')
    parameter_file = LaunchConfiguration('params_file')
    node_name = 'ouster_driver'

    # Acquire the driver param file
    params_declare = DeclareLaunchArgument('params_file',
                                           default_value=os.path.join(
                                               share_dir, 'config', 'ouster_fl_config.yaml'),
                                           description='FPath to the ROS2 parameters file to use.')

    driver_node = LifecycleNode(package='ros2_ouster',
                                executable='ouster_driver',
                                name=node_name,
                                output='screen',
                                emulate_tty=True,
                                parameters=[parameter_file],
                                arguments=['--ros-args', '--log-level', 'INFO'],
                                namespace='',
                                )

    configure_event = EmitEvent(
        event=ChangeState(
            lifecycle_node_matcher=matches_action(driver_node),
            transition_id=lifecycle_msgs.msg.Transition.TRANSITION_CONFIGURE,
        )
    )

    activate_event = RegisterEventHandler(
        OnStateTransition(
            target_lifecycle_node=driver_node, goal_state='inactive',
            entities=[
                LogInfo(
                    msg="[LifecycleLaunch] Ouster driver node is activating."),
                EmitEvent(event=ChangeState(
                    lifecycle_node_matcher=matches_action(driver_node),
                    transition_id=lifecycle_msgs.msg.Transition.TRANSITION_ACTIVATE,
                )),
            ],
        )
    )

    # TODO make lifecycle transition to shutdown before SIGINT
    shutdown_event = RegisterEventHandler(
        OnShutdown(
            on_shutdown=[
                EmitEvent(event=ChangeState(
                  lifecycle_node_matcher=matches_node_name(node_name=node_name),
                  transition_id=lifecycle_msgs.msg.Transition.TRANSITION_ACTIVE_SHUTDOWN,
                )),
                LogInfo(
                    msg="[LifecycleLaunch] Ouster driver node is exiting."),
            ],
        )
    )

    return LaunchDescription([
        params_declare,
        driver_node,
        activate_event,
        configure_event,
        shutdown_event,
    ])

```

### behavior
when there was only one LiDAR, The pointcloud topic bandwidth was normal as well as the rate of topic. When I plugged another LiDAR, the bandwidth increased and shown as the pic below.

[image]

[image]

SteveMacenski, 2023-05-10:
What's the ask / action? If there's that much data coming over the network, I'm not sure what you expect the ROS driver to do about it (e.g. we're not the ouster client code getting the packets nor the ROS standards for how pointclouds are published to end users)

ahuazuipiaoliang, 2023-05-11:
I am sorry, let me explain.
The bandwidth with was the bandwidth of pointcloud topic. I have multiple LiDARs OS LiDARs in this system, when there was only one LiDAR, the data was normal, but when I plugged another LiDAR, the pointcloud data went crazy. I think it something wrong with the pack parsing process.

ahuazuipiaoliang, 2023-05-11:
@SteveMacenski These LiDARs have different IP address. Have you tried multiple LiDARs?