ros-drivers/ros2_ouster_drivers #146: [Humble] Buildfarm error: JsonCpp::JsonCpp already exists

audrow, 2024-08-22:
I'm seeing this error on the buildfarm on Humble. It seems that ROS 2 Ouster hasn't changed on Humble since February, so I take it that an upstream repo has changed.

https://build.ros2.org/view/Hbin_uJ64/job/Hbin_uJ64__ros2_ouster__ubuntu_jammy_amd64__binary/89/

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/jsoncpp/jsoncpp-namespaced-targets.cmake:5 (add_library):
  add_library cannot create imported target "JsonCpp::JsonCpp" because
  another target with the same name already exists.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/jsoncpp/jsoncppConfig.cmake:41 (include)
  CMakeLists.txt:24 (find_package)
```

tfoote, 2024-08-22:
As found by Steven! this looks to be an issue upstream in jsoncpp https://github.com/open-source-parsers/jsoncpp/issues/1356 

It seems that the packaged version in debian/ubuntu has the effected version Jammy and newer: https://packages.ubuntu.com/search?keywords=jsoncpp&searchon=names/1000 and bookworm and newer: https://packages.debian.org/search?searchon=names&keywords=libjsoncpp

I don't see the debian packaging tracker aware of this: https://bugs.debian.org/cgi-bin/pkgreport.cgi?pkg=libjsoncpp-dev;dist=unstable  @roehling appears to be the maintainer.

roehling, 2024-08-22:
There is an open pull request to fix the issue: open-source-parsers/jsoncpp#1491
I backported that fix in Debian release 1.9.5-5; looking at the Ubuntu versions this means that jammy is affected but noble are not.

audrow, 2024-08-22:
Not a proper solution, but this might be an easy fix:

https://github.com/microsoft/vcpkg/pull/40548/files

tfoote, 2024-08-22:
Thanks for the info @roehling and getting that patch in promptly. At least we're only dealing with the older distros.

That explains why we're seeing an issue only in Humble. Presumably this has now regressed because a dependency has added jsoncpp as a dependency and that's why we're hitting the double declaration And with Ubuntu's release policy it's unlikely that we can get a backport for jammy

It looks like others have worked around this with a conditional around find_package(jsoncpp) such as trajopt: https://github.com/tesseract-robotics/trajopt/pull/288/files#diff-03702408a8d57132c3bcc29cbc672c295570a142a773ee85b4f78279f16fdfa3R16-R18 For getting this compiling that will likely be the best solution.

SteveMacenski, 2024-08-22:
What's the timeline on a fix? If we're talking about a week or two, we can just leave it. If its months, then probably worth a patch and re-release